MadLittleMods/node-usb-detection #6: Npm install failed on windows 8.1.

crapthings, 2015-02-13:
vs2013 express
python 2.7.x
nodejs 0.12
node gyp upgraded with npm install -g node-gyp

C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master>node-gyp rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@1.0.2
gyp info using node@0.12.0 | win32 | x64
child_process: customFds option is deprecated, use stdio instead.
gyp info spawn python
gyp info spawn args [ 'C:\\Users\\crapthings\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=auto',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\crapthings\\Downloads\\node-usb-detection-master\\node-usb-detection-master\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\crapthings\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\crapthings\\.node-gyp\\0.12.0\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\crapthings\\.node-gyp\\0.12.0',
gyp info spawn args   '-Dmodule_root_dir=C:\\Users\\crapthings\\Downloads\\node-usb-detection-master\\node-usb-detection-master',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'C:\\Users\\crapthings\\Downloads\\node-usb-detection-master\\node-usb-detection-master\\build',
gyp info spawn args   '-Goutput_dir=.' ]
gyp info spawn C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe
gyp info spawn args [ 'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64' ]
在此解决方案中一次生成一个项目。若要启用并行生成，请添加“/m”开关。
  detection.cpp
  deviceList.cpp
  detection_win.cpp
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xlocale(337): warning C4530: C++ exception handler used, but unwind semantics are not enabled. Specify /EHsc (..\src\deviceList.cpp) [C:
\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xlocale(337): warning C4530: C++ exception handler used, but unwind semantics are not enabled. Specify /EHsc (..\src\detection.cpp) [C:\
Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(15): error C2039: 'Arguments' : is not a member of 'v8' (..\src\detection.cpp) [C:\Users\crapthings\D
ownloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(15): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (..\sr
c\detection.cpp) [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(15): error C2143: syntax error : missing ',' before '&' (..\src\detection.cpp) [C:\Users\crapthings\D
ownloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(16): error C2065: 'uv_work_t' : undeclared identifier (..\src\detection.cpp) [C:\Users\crapthings\Dow
nloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(16): error C2065: 'req' : undeclared identifier (..\src\detection.cpp) [C:\Users\crapthings\Downloads
\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(16): error C2182: 'EIO_Find' : illegal use of type 'void' (..\src\detection.cpp) [C:\Users\crapthings
\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(17): error C2065: 'uv_work_t' : undeclared identifier (..\src\detection.cpp) [C:\Users\crapthings\Dow
nloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(17): error C2065: 'req' : undeclared identifier (..\src\detection.cpp) [C:\Users\crapthings\Downloads
\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(17): error C2182: 'EIO_AfterFind' : illegal use of type 'void' (..\src\detection.cpp) [C:\Users\crapt
hings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(19): error C2039: 'Arguments' : is not a member of 'v8' (..\src\detection.cpp) [C:\Users\crapthings\D
ownloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(19): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int (..\sr
c\detection.cpp) [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(19): error C2143: syntax error : missing ',' before '&' (..\src\detection.cpp) [C:\Users\crapthings\D
ownloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
c:\users\crapthings\downloads\node-usb-detection-master\node-usb-detection-master\src\detection.h(21): error C2039: 'Arguments' : is not a member of 'v8' (..\src\detection.cpp) [C:\Users\crapthings\D
[... 172 lines cut ...]
          C:\Users\crapthings\.node-gyp\0.12.0\deps\v8\include\v8.h(1599) : see declaration of 'v8::String'
          C:\Users\crapthings\.node-gyp\0.12.0\deps\v8\include\v8.h(1599) : see declaration of 'v8::String'
..\src\detection.cpp(102): error C3861: 'New': identifier not found [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
..\src\detection.cpp(103): error C2039: 'New' : is not a member of 'v8::String' [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
          C:\Users\crapthings\.node-gyp\0.12.0\deps\v8\include\v8.h(1599) : see declaration of 'v8::String'
          C:\Users\crapthings\.node-gyp\0.12.0\deps\v8\include\v8.h(1599) : see declaration of 'v8::String'
..\src\detection.cpp(103): error C3861: 'New': identifier not found [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
..\src\detection.cpp(103): fatal error C1003: error count exceeds 100; stopping compilation [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\include\xlocale(337): warning C4530: C++ exception handler used, but unwind semantics are not enabled. Specify /EHsc (..\src\detection_win.cpp)
[C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\build\detection.vcxproj]
..\src\detection_win.cpp(5): fatal error C1083: Cannot open include file: 'atlstr.h': No such file or directory [C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master\buil
d\detection.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\12.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\crapthings\AppData\Roaming\npm\node_modules\node-gyp\lib\build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Windows_NT 6.3.9600
gyp ERR! command "node" "C:\\Users\\crapthings\\AppData\\Roaming\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok

C:\Users\crapthings\Downloads\node-usb-detection-master\node-usb-detection-master>

MadLittleMods, 2015-05-30:
:+1: 

`npm install usb-detection` fails on Windows 8.1

Here is [my cli error output](http://pastebin.com/5DVzzYuA)

adrai, 2015-05-30:
This module is not compatible with node 0.12 :-(

MadLittleMods, 2015-05-30:
@adrai I am attempting to upgrade it.

Currently stuck on:

`declaration.cpp`, `NotifyAdded`
```
v8::Function::Cast(*addedCallback)->Call(isolate->GetCurrentContext()->Global(), 1, argv);
```

```
..\src\detection.cpp(67): error C2248: 'v8::Persistent<v8::Function,v8::NonCopyablePersistentTraits<T>>::operator *' : cannot acces
s private member declared in class 'v8::Persistent<v8::Function,v8::NonCopyablePersistentTraits<T>>' [C:\Users\madli_000\Downloads\
node-usb-detection-master\build\detection.vcxproj]
          with
          [
              T=v8::Function
          ]
          C:\Users\madli_000\.node-gyp\2.0.1\deps\v8\include\v8.h(762) : see declaration of 'v8::Persistent<v8::Function,v8::NonCop
  yablePersistentTraits<T>>::operator *'
          with
          [
              T=v8::Function
          ]
```

adrai, 2015-05-30:
I suggest to use: https://github.com/nodejs/nan

MadLittleMods, 2015-05-30:
I will tackle this in the morning.

It looks like [`NanCallback`](https://github.com/nodejs/nan#nancallback) is what I need but I still get the same error when using:

```
NanCallback *callback = new NanCallback(addedCallback);
callback->Call(1, argv);
```

MadLittleMods, 2015-05-31:
I have figured out most of the transformations for the Node v0.12.x but I am stuck on [this line](https://github.com/KABA-CCEAC/node-usb-detection/blob/master/src/detection.cpp#L169).

I have tried the following but it results in an error and while I have done similar cast/conversion before, I can't make it work here.

```
v8::Local<v8::Function> callback;
callback = args[0];
// ...

// Line 169: https://github.com/KABA-CCEAC/node-usb-detection/blob/master/src/detection.cpp#L169
ListBaton* baton = new ListBaton();
strcpy(baton->errorString, "");
baton->callback = new NanCallback(callback);
```

```
..\src\detection.cpp(183): error C2248: 'v8::Persistent<v8::Value,v8::NonCopyablePersistentTraits<T>>::Persistent' : cannot acce
ss private member declared in class 'v8::Persistent<v8::Value,v8::NonCopyablePersistentTraits<T>>' [C:\Users\madli_000\Downloads
\node-usb-detection-master\build\detection.vcxproj]
          with
          [
              T=v8::Value
          ]
          C:\Users\madli_000\.node-gyp\2.0.1\deps\v8\include\v8.h(761) : see declaration of 'v8::Persistent<v8::Value,v8::NonCop
  yablePersistentTraits<T>>::Persistent'
          with
          [
              T=v8::Value
          ]
```

MadLittleMods, 2015-05-31:
I made pull #9 that solves this issue.

closed 2015-05-31 by adrai