MadLittleMods/node-usb-detection #7: npm install fails on unbuntu 14

HERRKIN, 2015-03-25:
gyp ERR! node -v v0.10.37
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok 

npm ERR! usb-detection@1.0.3 preinstall: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb-detection@1.0.3 preinstall script.
npm ERR! This is most likely a problem with the usb-detection package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls usb-detection
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.13.0-48-generic

skilbjo, 2015-04-09:
+1 

Fails with me too, OS X 10.10.2, although this may be a node-gyp error rather than a usb-detection error

````
$ which node && node -v
/usr/local/bin/node
v0.12.2
$ which npm && npm -v
/usr/local/bin/npm
2.7.5
````
and
````
Package dbus-1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `dbus-1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'dbus-1' found
gyp: Call to 'pkg-config dbus-1 --libs' returned exit status 1. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:343:16)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1074:12)
gyp ERR! System Darwin 14.1.0
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/skilbjo/Code/Node/dropkey/node_modules/node-dbus
gyp ERR! node -v v0.12.2
gyp ERR! node-gyp -v v1.0.3
gyp ERR! not ok 
npm ERR! Darwin 14.1.0
npm ERR! argv "node" "/usr/local/bin/npm" "install"
npm ERR! node v0.12.2
npm ERR! npm  v2.7.5
npm ERR! code ELIFECYCLE

npm ERR! usb-detection@1.0.3 preinstall: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb-detection@1.0.3 preinstall script 'node-gyp rebuild'.
````

adrai, 2015-05-31:
try new version

MadLittleMods, 2015-06-09:
The latest v1.1.0 release adds Node v0.12.x support.

Using the new version, v1.1.0, as adrai suggested should solve your problem.

closed 2015-06-15 by MadLittleMods