MadLittleMods/node-usb-detection #8: deviceAddress does not match libusb device address

matthova, 2015-04-29:
The deviceAddress provided by usb-detection is not consistent with the same device's device address via libusb. This is on Windows 7, Node 0.10.35.

For example
usb-detection provides:
inDevice
 { locationId: 0,
  vendorId: 1133,
  productId: 50484,
  deviceName: 'USB Composite Device',
  manufacturer: '(Standard USB Host Controller)',
  serialNumber: '',
  deviceAddress: 18 }

while (npm's) usb provides
 { busNumber: 1,
  deviceAddress: 21,
  deviceDescriptor:
   { bLength: 18,
     bDescriptorType: 1,
     bcdUSB: 512,
     bDeviceClass: 0,
     bDeviceSubClass: 0,
     bDeviceProtocol: 0,
     bMaxPacketSize0: 8,
     idVendor: 1133,
     idProduct: 50484,
     bcdDevice: 10496,
     iManufacturer: 1,
     iProduct: 2,
     iSerialNumber: 0,
     bNumConfigurations: 1 },
  portNumbers: [ 1 ] }

The offset between libusb and usb-detection changes with any instance of the node process.

All the same, awesome library! Very useful.

MadLittleMods, 2015-06-15:
Thank you for the bug report. I saw your issue the moment you posted but I haven't gotten around to investigating.

thegecko, 2020-05-26:
Just hit this issue, any idea where the problem may be?

thegecko, 2020-08-31:
On Windows it looks like it's just incrementing an internal counter which makes it unique (to this library) but certainly isn't correct:

https://github.com/MadLittleMods/node-usb-detection/blob/d87c4669c5fcfd4c4d963892f17c1ba74456beb8/src/detection_win.cpp#L424