MadLittleMods/node-usb-detection #10: how can I get device path?

masnatacion, 2015-06-04:
do u know how can I get device path?

I only have:

locationId: 337641472,
vendorId: 6309,
productId: 770,
deviceName: 'STORE N GO',
manufacturer: 'Verbatim',
serialNumber: 'C400446A92C22022',
deviceAddress: 11

masnatacion, 2015-06-04:
I will want to save a file in my STORE  USB, so will want to know what is the path ?

MadLittleMods, 2015-06-04:
You can use the [`usb`](https://www.npmjs.com/package/usb) package.

```
usb.findByIds(vid, pid)
```

masnatacion, 2015-06-10:
thanks , i ge it this with usb module:

{ busNumber: 20,
  deviceAddress: 11,
  deviceDescriptor:
   { bLength: 18,
     bDescriptorType: 1,
     bcdUSB: 512,
     bDeviceClass: 0,
     bDeviceSubClass: 0,
     bDeviceProtocol: 0,
     bMaxPacketSize0: 64,
     idVendor: 6309,
     idProduct: 770,
     bcdDevice: 256,
     iManufacturer: 1,
     iProduct: 2,
     iSerialNumber: 3,
     bNumConfigurations: 1 },
  portNumbers: [ 1 ] }

so, where is the path?

masnatacion, 2015-06-10:
i wan to do this in a STORE USB:

var fs = require('fs');
var wstream = fs.createWriteStream('myOutput.txt');
wstream.write('Hello world!\n');
wstream.write('Another line');
// on Node.js older than 0.10, add cb to end()
wstream.end(function () { console.log('done'); });

masnatacion, 2015-06-11:
I will run this bat when some usb detected, its not the best solution but worked :(

@echo off

for /F "usebackq tokens=1,2,3,4 " %%i in (`wmic logicaldisk get caption^,description^,drivetype 2^>NUL`) do (

if %%l equ 2 (
			echo %%i
			xcopy /s "1.txt" %%i\
        )
)

MadLittleMods, 2015-06-11:
@masnatacion Stay tuned to whatever happens in #11.

closed 2015-09-19 by MadLittleMods

sudhakar3697, 2019-09-24:
A hack to get the mount path
https://gist.github.com/sudhakar3697/e065128a7ca5694647aac7b94f481289

muildrik, 2019-11-28:
No, you cannot get the mount path with just this module. However, this module can be perfectly combined with the drivelist module. I’ve tweaked that so it also provides me with the serial number. I then use the usb-detection module to get the serial number, vendor and product ids (and a bunch of other info), and those together then provide enough information to link the data from these three modules (drivelist, usb-detection, usb) together and get a full list of properties for your peripherals