MadLittleMods/node-usb-detection #12: Can't install 1.1.0 on RapsberryPi2 B

guillaumewuip, 2015-07-07:
Hi, I'm currently trying to usb `usb-detection` on my RaspberryPi2 B with node `0.10.26` and npm `1.4.28`.

The command `npm install usb-detection` or `npm install usb-detection@1.1.0`  failed with this error :

```
> usb-detection@1.1.0 install /home/pi/panoPi/node_modules/usb-detection
> node-gyp rebuild
make: Entering directory '/home/pi/panoPi/node_modules/usb-detection/build'
  CXX(target) Release/obj.target/detection/src/detection.o
In file included from ../src/detection.cpp:1:0:
../src/detection.h:17:17: error: ‘FunctionCallbackInfo’ in namespace ‘v8’ does not name a type
../src/detection.h:17:21: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/detection.h:17:41: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/detection.h:21:28: error: ‘FunctionCallbackInfo’ in namespace ‘v8’ does not name a type
../src/detection.h:21:32: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/detection.h:21:52: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/detection.h:23:27: error: ‘FunctionCallbackInfo’ in namespace ‘v8’ does not name a type
../src/detection.h:23:31: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/detection.h:23:51: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/detection.h:38:26: error: ‘FunctionCallbackInfo’ in namespace ‘v8’ does not name a type
../src/detection.h:38:30: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/detection.h:38:50: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/detection.h:40:28: error: ‘FunctionCallbackInfo’ in namespace ‘v8’ does not name a type
../src/detection.h:40:32: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/detection.h:40:52: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/detection.cpp:19:26: error: ‘FunctionCallbackInfo’ in namespace ‘v8’ does not name a type
../src/detection.cpp:19:30: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/detection.cpp:19:50: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/detection.cpp: In function ‘void RegisterAdded(int)’:
../src/detection.cpp:22:32: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../src/detection.cpp:22:32: note: candidates are:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../src/detection.cpp:26:7: error: ‘args’ was not declared in this scope
../src/detection.cpp:28:65: error: return-statement with a value, in function returning 'void' [-fpermissive]
../src/detection.cpp:31:7: error: ‘args’ was not declared in this scope
../src/detection.cpp:36:67: error: return-statement with a value, in function returning 'void' [-fpermissive]
../src/detection.cpp:39:39: error: expected primary-expression before ‘>’ token
../src/detection.cpp:39:41: error: expected primary-expression before ‘)’ token
../src/detection.cpp: In function ‘void NotifyAdded(ListResultItem_t*)’:
../src/detection.cpp:49:32: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../src/detection.cpp:49:32: note: candidates are:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
../src/detection.cpp:54:57: error: no matching function for call to ‘v8::Object::New(v8::Isolate*&)’
../src/detection.cpp:54:57: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1728:33: note: static v8::Local<v8::Object> v8::Object::New()
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1728:33: note:   candidate expects 0 arguments, 1 provided
../src/detection.cpp:55:15: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:55:113: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, int&)’
../src/detection.cpp:55:113: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note:   candidate expects 1 argument, 2 provided
../src/detection.cpp:56:15: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:56:109: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, int&)’
../src/detection.cpp:56:109: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note:   candidate expects 1 argument, 2 provided
../src/detection.cpp:57:15: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:57:111: error: no matching function for call to ‘v8::Number::New(v8::Isolate*&, int&)’
../src/detection.cpp:57:111: note: candidate is:
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note: static v8::Local<v8::Number> v8::Number::New(double)
/home/pi/.node-gyp/0.10.36/deps/v8/include/v8.h:1381:33: note:   candidate expects 1 argument, 2 provided
../src/detection.cpp:58:15: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:58:74: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:59:15: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:59:75: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/detection.cpp:60:15: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
[... 310 lines cut ...]
npm ERR! not ok code 0
```

Whereas `@1.0.0-1.0.3` install itself fine.

```
> usb-detection@1.0.3 preinstall /home/pi/panoPi/node_modules/usb-detection
> node-gyp rebuild

make: Entering directory '/home/pi/panoPi/node_modules/usb-detection/build'
  CXX(target) Release/obj.target/detection/src/detection.o
  CXX(target) Release/obj.target/detection/src/deviceList.o
  CXX(target) Release/obj.target/detection/src/detection_linux.o
  SOLINK_MODULE(target) Release/obj.target/detection.node
  SOLINK_MODULE(target) Release/obj.target/detection.node: Finished
  COPY Release/detection.node
make: Leaving directory '/home/pi/panoPi/node_modules/usb-detection/build'
usb-detection@1.0.3 node_modules/usb-detection
├── bindings@1.1.0
└── eventemitter2@0.4.14
```

Is node `0.10.36` no more supported ?

Thanks :)

MadLittleMods, 2015-07-07:
@guillaumewuip We are now using [`nan`](https://www.npmjs.com/package/nan) to abstract node API layer. I am under the impression that those calls are compatible with "Node 0.8->0.12" as stated in the `nan` readme.

Just some notes, I haven't delved into this issue yet.

guillaumewuip, 2015-07-07:
Ok thanks @MadLittleMods, I haven't really the knowledge to help debug this but feel free to ask me logs or anything.

tommy7600, 2015-10-06:
Try to use gcc 4.8 and g++ 4.8

tommy7600, 2015-10-14:
With some devices I get infinite loop with add and remove events. On windows and mac os everything is ok.

apla, 2016-03-15:
I've added patch for node 0.10.x #26

MadLittleMods, 2017-03-14:
@guillaumewuip Are things working for you now in 1.4.0 with @apla's change?

guillaumewuip, 2017-03-14:
@MadLittleMods I'm sorry to not know. I don't even remember for which project I try to use `usb-detection`. I can't test it now, sorry :/

closed 2017-03-15 by MadLittleMods