MadLittleMods/node-usb-detection #13: atlstr.h Not Found with VS2012 error

gauthamn28, 2015-08-11:
I saw that you had added support for Node.js v0.12.x. I currently have node.js version - v0.12.4 and node-gyp - 2.0.2 

I am getting the following error when installing the module - 

..\src\detection_win.cpp(5): fatal error C1083: Cannot open include file: 'atlstr.h': No such file or directory [C:\Users\gauthamn\node_modules\usb-detection\build\detection.vcxproj]

MadLittleMods, 2015-08-11:
From some googling, `atlstr.h` is part of the ATL library which gets included with some VS installs. Which version of VS exactly do you have?

In my VS installs I can find it in these places:

 - `C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\atlmfc\include\atlstr.h`

Here are some links that helped me investigate your issue:
 - http://stackoverflow.com/a/16404536/796832
 - https://groups.google.com/forum/#!topic/microsoft.public.vc.language/pUU-NM4RRhA

closed 2015-08-12 by MadLittleMods

gauthamn28, 2015-08-12:
Thanks for the help. The problem was that I had the express version but I
needed the Complete version for it to work.

MadLittleMods, 2015-08-12:
@gauthamn28 Thank you for the follow-up :)