MadLittleMods/node-usb-detection #15: Build fails with Node 4.x and NAN 2.x

todbot, 2015-09-30:
Trying to use 'usb-detection' with Node 4 fails, because of NAN 2.x.
To test:
1. Install Node 4.x
2. npm install usb-detection
3. See errors because of NAN API changes

(btw, thank you for usb-detection, it works great on nan <2)

todbot, 2015-09-30:
Oh and for hints on how to convert to nan v2, I've been looking at: https://github.com/lorenc-tomasz/node-hid/commit/50de11d5a28b44a2f6a21e9c215cb839ff92e04f

MadLittleMods, 2015-10-03:
:( I was hoping `nan` would smooth over all of these inconsistencies

MadLittleMods, 2015-10-12:
@todbot We now have support thanks to #16. I released a [new version `v1.3.0` on npm](https://www.npmjs.com/package/usb-detection)

closed 2015-10-12 by MadLittleMods

todbot, 2015-10-12:
Awesome, thanks Eric!