MadLittleMods/node-usb-detection #19: Mount point of connected device

skyksandr, 2015-11-02:
Hi, 

Is there any way to get mount point of connected device?

MadLittleMods, 2015-11-03:
Related to:

 - https://github.com/MadLittleMods/node-usb-detection/issues/18
 - https://github.com/MadLittleMods/node-usb-detection/issues/10

skyksandr, 2015-11-03:
For OS X I think it can be possible with `kiobsdnamekey`.

```
CFStringRef bsdName = ( CFStringRef ) IORegistryEntrySearchCFProperty ( usbDevice,
                                                                       kIOServicePlane,
                                                                       CFSTR ( kIOBSDNameKey ),
                                                                       kCFAllocatorDefault,
                                                                       kIORegistryIterateRecursively );
```

But because I'm not familiar with C++ and IOKit I failed with make it work by just adding this code to `detection_mac.cpp`

skyksandr, 2015-11-11:
I made simple usb-mount-notifier if anyone interested to.
https://github.com/skyksandr/usb-mount-notifier/tree/master

going to use it this way:
```
var spawn = require('child_process').spawn,
    usb_notifier = spawn('usb-mount-notifier');

usb_notifier.stdout.on('data', function (data) {
  console.log('stdout: ' + data);
});
```

sboudouk, 2015-11-23:
Can you make a simple README where you explain how to install it @skyksandr ? Does this work for Linux / Windows ?

skyksandr, 2015-11-23:
@sboudouk Nope, it doesn't. But I figured out how to handle this on windows. You can find 'copy-pastable' solution here:
https://github.com/skyderby/skyderby_desktop/blob/master/app/usb-mount-notify.js
Windows part a bit complicated but it is it simplest solution I think.

To install `usb-mount-notifier` just copy binary from `build` directory to your project.

sboudouk, 2015-11-23:
Thanks for your answer @skyksandr , i starred you by the way.

sudhakar3697, 2019-09-24:
https://gist.github.com/sudhakar3697/e065128a7ca5694647aac7b94f481289