MadLittleMods/node-usb-detection #23: does not exit event after stopMonitoring

openhoat, 2015-12-31:
Hi !

This code sample does not exit :
require('usb-detection');

$ node -v
v0.12.9

The problem is in index.js / line 10 : var detection = require('bindings')('detection.node');

MadLittleMods, 2016-01-02:
See https://github.com/MadLittleMods/node-usb-detection#the-scriptprocess-is-not-exitingquiting

closed 2016-01-02 by MadLittleMods

openhoat, 2016-01-03:
I know, I have read the doc and tested it :

require('usb-detection').usbDetect.stopMonitoring(function(){console.log('stopped');});
=> does not exit, and no callback invoked

MadLittleMods, 2016-01-03:
You are doing it wrong. `stopMonitoring` doesn't accept a callback, see: https://github.com/MadLittleMods/node-usb-detection/blob/4048b19279d32a78b83ad3237beaefd7854408bb/index.js#L97

Also `usbDetect` isn't a property of the `require('usb-detection')` module/package.

Here is the example from the readme:

```js
var usbDetect = require('usb-detection');

// Do some detection

// After this call, the process will be able to quit
usbDetect.stopMonitoring();
```

openhoat, 2016-01-03:
Ok, there was a copy/paste mistake in my last comment, even though the process does not exit (with or without callback this is the same).

Try this simple test and you will see it does not exit :
$ node -e "require('usb-detection').stopMonitoring()"
^C => have to stop it with Ctrl-C
$ node -v
v0.12.9

falnos24865, 2017-09-28:
Why is this closed? This is still a huge issue. Even using the stopMonitoring function in the docs the process fails to exit, requiring a third party hard kill on the process. Can this please be fixed? Its rendering this module almost useless if our software cant stop when its used.

MadLittleMods, 2017-10-12:
@falnos24865 Can you create a new issue with your OS and a small snippet that reproduced this now?

Could this actually be https://github.com/MadLittleMods/node-usb-detection/issues/2