MadLittleMods/node-usb-detection #25: Unable to require in node.js application

steviemo, 2016-02-18:
Hi folks,
I've added version 1.3.0 into my package.json but when I attempt to require in my app I get the following error:

build/node_modules/usb-detection/node_modules/bindings/binding…:83 Uncaught Error: Module version mismatch. Expected 47, got 46.

NPM version: 3.5.3
Node version: v4.2.4

Any ideas how I can use this module?

Thanks,
Stephen

MadLittleMods, 2016-02-18:
Did you recently upgrade to node v4 or npm 3? Are you using nvm? Were you using a previous version of `usb-detection` before installing the latest?

Perhaps you just need to run [`npm rebuild`](https://docs.npmjs.com/cli/rebuild)

steviemo, 2016-02-18:
Hi,

Thanks for the quick reply! I'm actually using https://github.com/docker/kitematic to get my head around node.js. I wasn't using a previous version of usb-detection. I'm using NPM 3 and node v4 not used the version manager.

steviemo, 2016-02-19:
Updated local development environment to node v4.3.1 (which brings in npm 2.14.12). Removed node_modules from root directory and ran npm install.

Still same error when trying to require usb-detection

bash-3.2$ npm list bindings
Kitematic@0.10.0 /Users/steviemo/workspaces/kitematic
└─┬ usb-detection@1.3.0
  └── bindings@1.1.0

steviemo, 2016-02-19:
Started fresh electron workspace using the quick start: https://github.com/atom/electron-quick-start

- Added usb-detection to dependencies
- main.js: `const usb = require('usb-detection');` 

`bash-3.2$ npm install
-
> usb-detection@1.3.0 install /Users/steviemo/Desktop/electron-quick-start/node_modules/usb-detection
> node-gyp rebuild

  CXX(target) Release/obj.target/detection/src/detection.o
  CXX(target) Release/obj.target/detection/src/deviceList.o
  CXX(target) Release/obj.target/detection/src/detection_mac.o
  SOLINK_MODULE(target) Release/detection.node

> usb-detection@1.3.0 postinstall /Users/steviemo/Desktop/electron-quick-start/node_modules/usb-detection
> node-gyp rebuild

  CXX(target) Release/obj.target/detection/src/detection.o
  CXX(target) Release/obj.target/detection/src/deviceList.o
  CXX(target) Release/obj.target/detection/src/detection_mac.o
  SOLINK_MODULE(target) Release/detection.node
usb-detection@1.3.0 node_modules/usb-detection
├── bindings@1.1.0
├── eventemitter2@0.4.14
├── bluebird@2.10.2
└── nan@2.2.0
bash-3.2$ npm start

> electron-quick-start@1.0.0 start /Users/steviemo/Desktop/electron-quick-start
> electron main.js

App threw an error when running [Error: Module version mismatch. Expected 47, got 46.]
`

steviemo, 2016-02-19:
Should have said `npm rebuild` doesn't seem to solve the issue either

steviemo, 2016-02-19:
OK a bit more debugging of the exception when putting a break point at 82 of bindings.js for exception e:

`Error: Cannot find module '/Users/steviemo/workspaces/kitematic/build/node_modules/usb-detection/build/detection.node' at Function.Module._resolveFilename (module.js:338:15) at Function.Module._load (module.js:289:25) at Module.require (module.js:366:17) at require (module.js:385:17) at bindings (/Users/steviemo/workspaces/kitematic/build/node_modules/usb-detection/node_modules/bindings/bindings.js:76:44) at Object.<anonymous> (/Users/steviemo/workspaces/kitematic/build/node_modules/usb-detection/index.js:10:37) at Module._compile (module.js:425:26) at Object.Module._extensions..js (module.js:432:10) at Module.load (module.js:356:32) at Function.Module._load (module.js:313:12) at Module.require (module.js:366:17) at require (module.js:385:17) at Object.<anonymous> (/Users/steviemo/workspaces/kitematic/build/app.js:59:21) at Module._compile (module.js:425:26) at Object.Module._extensions..js (module.js:432:10) at Module.load (module.js:356:32)`

And fair enough I'm not seeing a detection.node file at this location. Is there anything I can do on my end to workaround this Eric?

MadLittleMods, 2016-02-20:
Hey @steviemo,

I just tested `usb-detection` on OSX(npm@2.14.7 and node@4.2.3) and Windows(npm@3.7.1 and node@5.5.0) and it works completely fine for me.

steviemo, 2016-02-22:
Hi Eric,
I'm on OSX 10.11.3 and have XCode with command line tools installed.

```
bash-3.2$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/usr/include/c++/4.2.1
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.3.0
Thread model: posix
```

I didn't see any errors when installing usb-detection, I attached the CLI in comment number 4. I only see the module version error once I try to start a existing or new Electron application. Perhaps this are conflicts with Electron? 

Should I try removing node from my system and starting fresh? I'm new to Node - so if you have any suggestions that would be great.

steviemo, 2016-02-22:
Just cloned this repo and did an install then npm test and adding/removing UDB device seems to work OK. Must be an issue with using this library within an Electron application

MadLittleMods, 2016-02-22:
Can you trying cloning this package and running the tests outside of Electron? Also try making a small demo outside of electron. This will narrow it down to ether it is an Electron problem

steviemo, 2016-02-22:
Aye that's what I tried and tested passed without any issue. Must be something specific with using this module with Electron

MadLittleMods, 2016-02-22:
Because `usb-detection` is a native module, you probably need to do this: https://github.com/atom/electron/blob/master/docs/tutorial/using-native-node-modules.md

For nw.js, https://github.com/nwjs/nw.js/wiki/build-native-modules-with-nw-gyp, https://github.com/nwjs/nw.js/wiki/using-node-modules#3rd-party-modules-with-cc-addons

steviemo, 2016-02-22:
I just came across this myself and was about to post!

I'm going to test this in the morning and let u know the results! Thanks again!

steviemo, 2016-02-23:
Got it working in electron using election rebuild. Thanks!!

closed 2016-02-23 by MadLittleMods

lalitmee, 2018-02-23:
Hey @steviemo , actually I am facing the same issue. can you help me. How you solved it by using electron rebuilt?

steviemo, 2018-02-23:
I think I did `npm install --save-dev electron-rebuild` from https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md sorry it was in a previous company so cannot remember for sure as i don't have the code in front of me.

lalitmee, 2018-02-23:
Oh.. thank you so much for this fast reply. But only by installing, this will work or I have to do something in my code?

lalitmee, 2018-02-23:
Actually what I am facing is that, when I am trying to run command for webpack. There is no error but these two warnings are there.
```
WARNING in ./node_modules/bindings/bindings.js
76:22-40 Critical dependency: the request of a dependency is an expression
 @ ./node_modules/bindings/bindings.js
 @ ./node_modules/usb-detection/index.js
 @ ./app/app.js

WARNING in ./node_modules/bindings/bindings.js
76:43-53 Critical dependency: the request of a dependency is an expression
 @ ./node_modules/bindings/bindings.js
 @ ./node_modules/usb-detection/index.js
 @ ./app/app.js
```
I am not getting why these are occuring.

ivankennethwang, 2018-09-19:
@lalitmee did you managed to make it work? Can you share your steps?

lalitmee, 2018-09-19:
@ivankennethwang, actually it was a long back when I was facing this issue. So I don't remember completely but as far as I remember there was something related to installing electron locally or globally. :innocent: