MadLittleMods/node-usb-detection #27: Problem packaging for windows

primozs, 2016-06-12:
Did anybody have similar problem while running packaged electron app for windows?
I have tried to packaged it on mac and on windows but the error is the same.
```
win32-ia32\resources\app\node_modules\bindings\bindings…:83 Uncaught Error: %1 is not a valid Win32 application.
win32-ia32\resources\app\node_modules\usb-detection\build\Release\detection.node
```

MadLittleMods, 2016-06-13:
Have you gone through this? https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md

primozs, 2016-06-13:
@MadLittleMods Hello, I did run ```electron-rebuild``` but maybe not on the production build. Thank you for the tip. I will check it out in more detail.
Cheers

primozs, 2016-06-14:
@MadLittleMods neither manual linking or electron-rebuild did not help the only way for me was to build it on windows but I had to install visual studio and everything.
Thank you, cheers

closed 2016-06-14 by primozs