MadLittleMods/node-usb-detection #28: Should include Start Monitoring in ReadMe

andrewpe, 2016-06-30:
Considering stop monitoring is in the readme.  It would be helpful if the start monitoring function call was explained in the readme as well.  Even though it should be obvious, it would be clearer than common sense or to look through the code.

MadLittleMods, 2016-07-01:
Feel free to make a PR to add `startMonitoring` and `stopMonitoring` API docs to the readme.

MadLittleMods, 2017-03-14:
Docs on `stopMonitoring` were added in this commit, https://github.com/MadLittleMods/node-usb-detection/commit/dc61072ea5c38b25d0c406ad7fc99c60f6ca8664

See https://github.com/MadLittleMods/node-usb-detection#the-scriptprocess-is-not-exitingquiting

closed 2017-03-14 by MadLittleMods

Haraldson, 2017-07-02:
The API I’m currently getting is `{ find, registerAdded, registerRemoved, startMonitoring, stopMonitoring }`. The current docs does not make it clear how I’m supposed to listen for events – there is no `on`.

MadLittleMods, 2017-07-22:
@Haraldson The `on` etc, comes from [`eventemitter2`](https://www.npmjs.com/package/eventemitter2)