MadLittleMods/node-usb-detection #30: node v6.6.0 support?

wkentdag, 2016-10-07:
I'm running into install issues similar to those described in #13. I'm running Windows 10 via bootcamp on a Macbook Pro. I've installed the visual studio build tools, visual studio community 2015, python 2.7 (and added to PATH), node-gyp, etc. I'm guessing my build issues are probably related to my VS install, but I wanted to eliminate one potential point of failure before I go down that path: does this library support node v6.6.0?

wkentdag, 2016-10-07:
ps - specifically, I'm using VS 14.0. I had a look in `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\`, where the missing `atlbase.h` file is supposed to be, and found many similar-lookin files, but not the one in question :(

```
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include>dir
 Volume in drive C is BOOTCAMP
 Volume Serial Number is 0E99-ED38

 Directory of C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include

10/07/2016  11:57 AM    <DIR>          .
10/07/2016  11:57 AM    <DIR>          ..
06/01/2016  01:06 PM           485,495 agents.h
06/01/2016  01:07 PM            10,970 agile.h
06/01/2016  01:06 PM           123,902 algorithm
06/01/2016  01:06 PM            20,742 allocators
06/01/2016  01:06 PM            13,713 ammintrin.h
06/01/2016  01:06 PM           307,400 amp.h
06/01/2016  01:06 PM           158,787 amprt.h
06/01/2016  01:06 PM             6,001 amprt_exceptions.h
06/01/2016  01:06 PM           233,750 amp_graphics.h
06/01/2016  01:06 PM           110,341 amp_math.h
06/01/2016  01:06 PM           879,429 amp_short_vectors.h
06/01/2016  01:06 PM            30,748 armintr.h
06/01/2016  01:06 PM           567,680 arm_neon.h
06/01/2016  01:06 PM            13,571 array
06/01/2016  01:06 PM            33,523 atomic
06/01/2016  01:06 PM            16,327 bitset
06/01/2016  01:06 PM               238 cassert
06/01/2016  01:06 PM               294 ccomplex
06/01/2016  01:06 PM             1,093 cctype
06/01/2016  01:06 PM               522 cerrno
06/01/2016  01:06 PM             1,040 cfenv
06/01/2016  01:06 PM               284 cfloat
06/01/2016  01:06 PM            33,031 chrono
06/01/2016  01:06 PM               741 cinttypes
06/01/2016  01:06 PM               289 ciso646
10/07/2016  11:57 AM    <DIR>          cliext
06/01/2016  01:06 PM               289 climits
06/01/2016  01:06 PM               590 clocale
06/01/2016  01:06 PM            18,280 cmath
10/07/2016  11:57 AM    <DIR>          CodeAnalysis
06/01/2016  01:06 PM            22,218 codecvt
06/01/2016  01:06 PM            73,667 collection.h
06/01/2016  01:06 PM            13,478 comdef.h
06/01/2016  01:06 PM            79,449 comdefsp.h
06/01/2016  01:06 PM            27,221 comip.h
06/01/2016  01:06 PM            33,151 complex
06/01/2016  01:06 PM            59,893 comutil.h
06/01/2016  01:06 PM           241,609 concrt.h
06/01/2016  01:06 PM            94,981 concrtrm.h
06/01/2016  01:06 PM            15,699 ConcurrencySal.h
06/01/2016  01:06 PM            34,458 concurrent_priority_queue.h
06/01/2016  01:06 PM            30,641 concurrent_queue.h
06/01/2016  01:06 PM            66,250 concurrent_unordered_map.h
06/01/2016  01:06 PM            59,263 concurrent_unordered_set.h
06/01/2016  01:06 PM            82,992 concurrent_vector.h
06/01/2016  01:06 PM             4,522 condition_variable
06/01/2016  01:06 PM             1,721 crtdefs.h
06/09/2016  10:03 PM               581 crtversion.h
06/01/2016  01:06 PM               547 csetjmp
06/01/2016  01:06 PM               710 csignal
06/01/2016  01:06 PM               402 cstdarg
06/01/2016  01:06 PM               294 cstdbool
06/01/2016  01:06 PM               768 cstddef
06/01/2016  01:06 PM             2,041 cstdint
06/01/2016  01:06 PM             1,866 cstdio
06/01/2016  01:06 PM             1,475 cstdlib
06/01/2016  01:06 PM             1,002 cstring
06/01/2016  01:06 PM               309 ctgmath
06/01/2016  01:06 PM               775 ctime
[... 121 lines cut ...]
06/01/2016  01:06 PM            92,236 xlocale
06/01/2016  01:06 PM            14,786 xlocbuf
06/01/2016  01:06 PM             9,111 xlocinfo
06/01/2016  01:06 PM             6,181 xlocinfo.h
06/01/2016  01:06 PM             4,168 xlocmes
06/01/2016  01:06 PM            30,195 xlocmon
06/01/2016  01:06 PM            51,464 xlocnum
06/01/2016  01:06 PM            33,557 xloctime
06/08/2016  09:12 PM            23,963 xmemory
06/08/2016  09:12 PM            34,076 xmemory0
06/01/2016  01:06 PM            19,389 xmmintrin.h
06/01/2016  01:06 PM            19,263 xstddef
06/01/2016  01:06 PM            76,732 xstring
06/01/2016  01:06 PM             7,201 xtgmath.h
06/01/2016  01:06 PM             7,574 xtr1common
06/01/2016  01:06 PM            66,600 xtree
06/01/2016  01:06 PM           111,290 xutility
06/01/2016  01:06 PM            51,719 xxamp.h
06/01/2016  01:06 PM             4,961 xxamp_inl.h
06/01/2016  01:06 PM            50,281 xxatomic
06/01/2016  01:06 PM             2,538 ymath.h
06/01/2016  01:06 PM            23,728 yvals.h
             195 File(s)      7,951,518 bytes
               9 Dir(s)  63,881,945,088 bytes free
```

MadLittleMods, 2016-10-08:
Just tested with this setup and everything is working fine for me.

WIndows 10 with Visual Studio 2015 (`C:\Program Files (x86)\Microsoft Visual Studio 14.0`

 - ✔️ node@6.7.0, npm@3.10.3
 - ✔️ node@6.3.1, npm@3.10.3
 - ✔️ node@4.5.0, npm@2.15.9

---

For the issue you mentioned, https://github.com/MadLittleMods/node-usb-detection/issues/13, I can find `atlstr.h` located here: `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\atlmfc\include\atlstr.h`

 - Paste the full log of the install that fails. (Put it in a code fence and `<details>` block)
 - Try reinstalling Visual Studio and don't exclude anything from the install

wkentdag, 2016-10-10:
Thanks for taking the time to test and respond. Unfortunately even after wiping VS and reinstalling I'm still running into install errors (albeit different ones):

First of all, the `atlmfc` subdir inside the VS program files is completely missing:

```
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC>dir
 Volume in drive C is BOOTCAMP
 Volume Serial Number is 0E99-ED38

 Directory of C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC

10/10/2016  12:35 PM    <DIR>          .
10/10/2016  12:35 PM    <DIR>          ..
10/10/2016  12:30 PM    <DIR>          bin
10/10/2016  12:18 PM    <DIR>          include
10/10/2016  12:18 PM    <DIR>          redist
10/10/2016  12:21 PM    <DIR>          Snippets
10/10/2016  12:35 PM    <DIR>          UnitTest
10/10/2016  12:34 PM    <DIR>          vcpackages
10/10/2016  10:37 AM    <DIR>          VCProjectDefaults
10/10/2016  12:18 PM    <DIR>          vcprojectitems
10/10/2016  12:21 PM    <DIR>          VCWizards
               0 File(s)              0 bytes
              11 Dir(s)  65,913,335,808 bytes free
```

In any case, I went ahead and tried to `npm install` anyway to see what would happen (after running `npm config set msvs_version 2015` per a related issue-thread [suggestion](https://github.com/felixrieseberg/windows-build-tools/issues/9#issuecomment-244793165)), and here's the output:

```
C:\Users\will\code\secure-key-fingerprint>npm install

> usb-detection@1.4.0 install C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection
> node-gyp rebuild

C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection>if not defined npm_config_node_gyp (node "C:\Users\will\AppData\Roaming\nvm\v6.6.0\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Platforms\x64\PlatformToolsets\v140\Toolset.targets(36,5): error
 MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or change the SDK vers
ion in the project property pages or by right-clicking the solution and selecting "Retarget solution". [C:\Users\will\c
ode\secure-key-fingerprint\node_modules\usb-detection\build\detection.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\will\AppData\Roaming\nvm\v6.6.0\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\will\\AppData\\Roaming\\nvm\\v6.6.0\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection
gyp ERR! node -v v6.6.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.6.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! usb-detection@1.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb-detection@1.4.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the usb-detection package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs usb-detection
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls usb-detection
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\will\code\secure-key-fingerprint\npm-debug.log
```

I still think that there must be something wrong with my VS install, although for the life of me can't figure out what. Here's a link to the [full output log](https://gist.github.com/wkentdag/df90c5c2046192a58470e129fff00146) from my VS install if that's helpful. Thanks again for your time.

MadLittleMods, 2016-10-10:
When I ran `npm config get msvs_version`, I got `2013` and looking at the programs installed, it is listed:

[image]

But if I do point it at my 2015 install, `npm config set msvs_version 2015`, it builds just fine as well.

----

Just to summarize that [log output you posted](https://github.com/MadLittleMods/node-usb-detection/issues/30#issuecomment-252732822):

>``` 
> MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK
> ```

That is available here, https://developer.microsoft.com/en-us/windows/downloads/windows-8-1-sdk

It looks like I may have it installed:

[image]

It's unclear where they are installed but perhaps:

[image]

---

From the issue you mentioned, have you tried:

> Workaround: Install Visual C++ projects support to Visual Studio 2015 (requires 3GB space).
>
>https://github.com/felixrieseberg/windows-build-tools/issues/9#issuecomment-252673089

---

What does your env path variable look like? `echo %PATH%`

Just for reference, installation instructions for `node-gyp`, https://github.com/nodejs/node-gyp#installation

wkentdag, 2016-10-10:
🎉Aha! I first installed the standalone Windows SDK with the link you provided, which produced a different error a little further along in the install: 

```
TRACKER : error TRK0005: Failed to locate: "CL.exe". The system cannot find the file specified. [C:\Users\will\code\sec
ure-key-fingerprint\node_modules\usb-detection\build\detection.vcxproj]

gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\will\AppData\Roaming\nvm\v6.6.0\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\will\\AppData\\Roaming\\nvm\\v6.6.0\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection
gyp ERR! node -v v6.6.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
```

I saw the comment you linked to in the other thread and *thought* that I was installing the C++ tools by un- and re-installing the entire VS community package. Thanks for clarifying that I actually needed to take the extra step of creating a new C++ project within VS - this finally prompted me to download the Visual C++ Build Tools, which includes the Windows 8.1 SDK - that step solved the issue. Why they don't make that package easier to find on the main downloads page is a mystery to me.

Really appreciate your tips and feedback on this issue, thanks! 🍻

wkentdag, 2016-10-11:
sorry, spoke too soon - still encountering install errors. I thought I successfully installed the module, as I wrote above, but didn't test the install until I ran my script, which is bundled into a windows service, and got the following error in my logs:

```
Error: Could not locate the bindings file. Tried:
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\build\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\build\Debug\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\build\Release\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\out\Debug\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\Debug\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\out\Release\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\Release\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\build\default\detection.node
 → C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\compiled\6.6.0\win32\x64\detection.node
    at bindings (C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\node_modules\bindings\bindings.js:88:9)
    at Object.<anonymous> (C:\Users\will\code\secure-key-fingerprint\node_modules\usb-detection\index.js:10:37)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (C:\Users\will\code\secure-key-fingerprint\app.js:1:75)
    at Module._compile (module.js:556:32)
    at Object.Module._extensions..js (module.js:565:10)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.runMain (module.js:590:10)
    at run (bootstrap_node.js:394:7)
```

None of the files it listed were there in my `node_modules`, but there is a `binding.gyp`. Thinking this might be a problem with my `node-gyp` installation, I rebuilt using the manual steps laid out in the README and wrote a little test script to isolate the problem, but I'm still running up against the following error:

```
C:\Users\will\code\usbtest>npm install

> usb-detection@1.4.0 install C:\Users\will\code\usbtest\node_modules\usb-detection
> node-gyp rebuild

C:\Users\will\code\usbtest\node_modules\usb-detection>if not defined npm_config_node_gyp (node "C:\Users\will\AppData\Roaming\nvm\v6.6.0\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  detection.cpp
  deviceList.cpp
  detection_win.cpp
  win_delay_load_hook.c

<details>

C:\Users\will\AppData\Roaming\nvm\v6.6.0\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.c(34): error C2373: '__pfnDliNotifyH
ook2': redefinition; different type modifiers [C:\Users\will\code\usbtest\node_modules\usb-detection\build\detection.vcxproj]

</details>

  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\delayimp.h(134): note: see declaration of '__pfnDliNotifyHook2'
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Users\will\AppData\Roaming\nvm\v6.6.0\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Windows_NT 10.0.14393
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\will\\AppData\\Roaming\\nvm\\v6.6.0\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\Users\will\code\usbtest\node_modules\usb-detection
gyp ERR! node -v v6.6.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
npm WARN usbtest@1.0.0 No description
npm WARN usbtest@1.0.0 No repository field.
npm ERR! Windows_NT 10.0.14393
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.6.0
npm ERR! npm  v3.10.3
npm ERR! code ELIFECYCLE

npm ERR! usb-detection@1.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb-detection@1.4.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the usb-detection package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs usb-detection
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls usb-detection
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\will\code\usbtest\npm-debug.log
```

MadLittleMods, 2016-10-13:
@wkentdag I can't reproduce and don't have any more insight than you. Please follow-up if you figure out what is going wrong.

wkentdag, 2016-10-13:
@MadLittleMods no worries, thanks for your insights thus far. will definitely post a followup if i can get this working 😄

Yadro, 2016-11-02:
I have same problem/

Yadro, 2016-11-02:
`npm install -g npm@next` fix problem

MadLittleMods, 2017-03-14:
@wkentdag Does [@Yadro's fix](https://github.com/MadLittleMods/node-usb-detection/issues/30#issuecomment-257983617) work for you?

wkentdag, 2017-03-14:
@MadLittleMods I ran into this problem during a freelance project and no longer have access to a Windows machine, so unfortunately I can't confirm :/

closed 2017-03-15 by MadLittleMods

timfish, 2017-05-23:
Thank you for this library!

I encountered this issue too. On my desktop the repository installed ok whereas on a fairly fresh Windows 10 laptop it failed because `atlstr.h` was missing.

On the laptop I only had installed Visual C++ Build Tools 2015 and the desktop has several full Visual Studio installs plus all the SDK's.

Installing Visual Studio 2015 Community with Window 10 SDK selected appears to have fixed it. So it looks like at least on fresh Windows 10, C++ Build Tools is not enough to build `node-usb-detection`!