MadLittleMods/node-usb-detection #31: Get Path from Devices

jdnichollsc, 2016-10-20:
Hi guys,

Is possible to get the path (file System) of each device? Any example about that?

Thanks in advance, Nicholls

MadLittleMods, 2016-10-20:
> Related to:
> 
>  - https://github.com/MadLittleMods/node-usb-detection/issues/18
>  - https://github.com/MadLittleMods/node-usb-detection/issues/10
>
> https://github.com/MadLittleMods/node-usb-detection/issues/19

MadLittleMods, 2016-10-21:
@jdnichollsc Do those references solve your use case?

Share your solution and close the issue accordingly

jdnichollsc, 2016-10-23:
I don't understand :/

MadLittleMods, 2016-10-24:
@jdnichollsc You gave a :+1: reaction to the reply with the references to other issues that tackle the same sort of problem. So I assume your figured out how to work something out. I was asking to share that solution you ended up with and if your issue is solved, close the issue.

closed 2016-10-24 by MadLittleMods

s-kashif, 2018-08-07:
Has anyone solved this issue yet?

jdnichollsc, 2018-08-08:
@MadLittleMods my 👍 reaction is for your response, but these other issues are not related with my problem, what's the way to get the devicePath?
Thanks, Nicholls

MadLittleMods, 2018-08-08:
@s-kashif @jdnichollsc There isn't a solution yet. the canonical issue for this is https://github.com/MadLittleMods/node-usb-detection/issues/19

jdnichollsc, 2018-08-08:
ohh ok, thanks for your time 👍

sudhakar3697, 2019-09-24:
A hack to get the mounted path
https://gist.github.com/sudhakar3697/e065128a7ca5694647aac7b94f481289