MadLittleMods/node-usb-detection #34: deviceName Garbled in windows

qqnp1100, 2016-11-18:
run in electron 1.4.5,
npm 4.0.2

log:
[image]
and in mac os，the  devicename is true.
Who can tell me how to solve it?

jyjunyz, 2018-08-21:
hi, has it been fixed yet？ I met the same problem

MadLittleMods, 2018-08-21:
@jyjunyz If there isn't an update in the issue, then there isn't an update.

MadLittleMods, 2018-12-06:
@marcong95 Can you provide the exact output from `usb-detection` for those devices? I'm curious what characters it is trying to use and mangling

MadLittleMods, 2018-12-06:
@marcong95 If you're interested in submiting a PR, the relevant code is here,

 - https://github.com/MadLittleMods/node-usb-detection/blob/875c525cc55b8c8d161138e5055c6b40a15d0851/src/detection_win.cpp#L422-L432
 - https://github.com/MadLittleMods/node-usb-detection/blob/875c525cc55b8c8d161138e5055c6b40a15d0851/src/detection.cpp#L185-L186

jyjunyz, 2018-12-07:
I resolved this, I will submit a PR later

marcong95, 2018-12-07:
@MadLittleMods I don't know too much about C++ and its toolchain, but I will give a try. If @jyjunyz submitted his PR, just ignore mine.

closed 2019-10-06 by MadLittleMods

MadLittleMods, 2019-10-06:
The [fix](https://github.com/MadLittleMods/node-usb-detection/pull/86) is now shipped in [`usb-detection@4.4.0`](https://github.com/MadLittleMods/node-usb-detection/blob/master/CHANGELOG.md#440---2019-10-6) :tada: 

Thanks to @nononoone for the [fix](https://github.com/MadLittleMods/node-usb-detection/pull/86) ❤️