MadLittleMods/node-usb-detection #35: Library makes `process.exit()` freeze

sebakerckhof, 2016-12-13:
I have a simple test script like:
```js
setTimeout(() => {
  process.exit(0);
},5000);
```
This works 100% of the times (after 5s the node process stops).

When I now include this library, like:
```js
const usbDetection = require('usb-detection');

usbDetection.on('add', device => { console.log('added', device); } );
usbDetection.on('remove', device => { console.log('removed', device) });

setTimeout(() => {
  process.exit(0);
},5000);
```

The process hangs most of the times after 5s instead of exiting.

sebakerckhof, 2016-12-13:
Seems related to: https://github.com/MadLittleMods/node-usb-detection/issues/2

MadLittleMods, 2016-12-13:
See https://github.com/MadLittleMods/node-usb-detection#the-scriptprocess-is-not-exitingquiting

falnos24865, 2017-09-28:
Not sure this is related to #2. I have applied to PR listed in #2 and it fixes the 100% CPU spike, but the usb-detection module still prevents node from existing once it is used, even after using the stopMonitoring function.

MadLittleMods, 2017-11-10:
@sebakerckhof @falnos24865 What OS are you on? I suspect you are on Linux after experiencing something similar now.

closed 2018-02-07 by MadLittleMods