MadLittleMods/node-usb-detection #36: serialNumber missing on Windows

mchristopher, 2017-01-05:
It looks like the USB binding code on Windows does not handle device serial numbers at all.

From what I can tell, the serial number must be extracted from the devicePath (format "USB\Vid_046d&Pid_c018\01" where "01" is the serial), but I don't have enough data to confirm that the format is consistent.

I'm happy to submit a PR against this project, but wanted to make sure that approach is correct before doing so.

ddombrowsky, 2018-02-15:
I think only mass storage devices have reliable serial numbers, but I can't decipher where in the device tree to read them.  I don't think the "01" in your example is correct.  In linux, the serial number is correctly presented by udev.  I don't know where it is in windows.  Still looking, as this is an essential part of my project.

todbot, 2018-02-15:
I think USB device descriptor serial number is just string index 2 (index 0 being manufacturer name, index 1 being product name). 
Here's how libusb does it for Windows:
https://github.com/libusb/libusb/blob/master/libusb/os/windows_winusb.c#L2782

ddombrowsky, 2018-02-15:
@todbot yes, that is how it's presented in libusb, but I don't know enough about the windows device path stuff to know how to get it to work in node-usb-detection, which does not depend on libusb.  I can't use node-usb/libusb because that requires "owning" the device.  I need the OS to mount it as normal, and I need to query the OS to get the information it read about the device.  

OP is correct, the device instance path contains the serial number in question.  I think it needs to be inserted in the code here:
```diff
diff --git a/src/detection_win.cpp b/src/detection_win.cpp
index e9a0362..fe99057 100644
--- a/src/detection_win.cpp
+++ b/src/detection_win.cpp
@@ -422,6 +424,13 @@ void ExtractDeviceInfo(HDEVINFO hDevInfo, SP_DEVINFO_DATA* pspDevInfoData, TCHAR
                // Use this to extract VID / PID
                extractVidPid(buf, resultItem);
        }
+
+       if (DllSetupDiGetDeviceRegistryProperty(hDevInfo, pspDevInfoData, SPDRP_LOCATION_PATHS, &DataT, (PBYTE)buf, buffSize, &nS
+            printf("YOYOY2 %s\n", buf);
+        }
+
+
+       resultItem->serialNumber = "whatever";
 }
```

But I can't find the magic api call to do it.  I'm probably missing something fundamental.

todbot, 2018-02-15:
Yes, I know little about how Windows USB stuff works and was throwing up stuff blindly.  If you could just get the entire USB device descriptor, the vendor name, product name, and serial numbers are *right there*. So frustrating.   In older libusb,  doing device descriptor requests was taxing on the system so your parsing of the device registry seems like a kinder way to go. I can't believe they don't put the serial numbers in there.

ddombrowsky, 2018-02-16:
This is windows.  Nothing is ever what it seems.  _waves hands_

ddombrowsky, 2018-02-17:
After spending all day on this, I realized I should have checked the pull requests first: https://github.com/MadLittleMods/node-usb-detection/pull/50, though it looks like that is a much more complex change than what I did.

The serial number for mass storage devices is already being fetched as part of the `DllSetupDiGetDeviceInstanceId` call.  The "dev instance" path contains the serial number.

MadLittleMods, 2018-02-17:
@mchristopher @ddombrowsky @todbot See https://github.com/MadLittleMods/node-usb-detection/pull/50. Feel free to pick it up and address the review 🙂 

Woah, you just found it 🎉

ddombrowsky, 2018-02-17:
Yep, it looks like we ended up doing much the same thing, @jasondana and I.

closed 2018-06-04 by MadLittleMods