MadLittleMods/node-usb-detection #39: Electron app with webpack using usb-detection

safinn, 2017-04-22:
Im trying to use `usb-detection` in an electron application. My electron application is using react and webpack. I'm on OSX with the command-line tools installed and this is in my webpack config:
```
loaders: [
      {
        test: /\.node$/,
        use: 'node-loader'
      }
]
```
and in my package.json I have this in scripts:
`"postinstall": "electron-rebuild --force"`

When I import this module and try and use it I get the error: 
`Uncaught Error: Could not locate the bindings file. Tried: ...`

Any ideas how I can overcome this? Thank you.

cpt-chewieeee, 2017-07-15:
I was able to figure this out and these are the steps I took:

1. remove node_modules
`rm -rf node_modules`

2. clean npm cache 
`npm cache clean`

I'm assuming you followed the two package.json structure....

3. `cd` into the directory with the second package.json and run
`npm install --save-dev electron-rebuilder && npm install --save usb-detection`

4. rebuild usb-detection with electron-rebuilder
`./node_modules/.bin/electron-rebuild [PATH_TO_USB_DETECTION]

My problem was that I install usb-detection and electron-rebuild under the first package.json where webpack lives.

BhaskarNewase, 2017-07-20:
How to use usb-detection with electron ?

How should i get data from usb-detection module to angular 2 ?

MadLittleMods, 2017-11-27:
Closing because of lack of response from OP.

Thanks for instructions @cpt-chewieeee ❤️

closed 2017-11-27 by MadLittleMods