MadLittleMods/node-usb-detection #40: Is it possible to distribute pre-built binaries for this module!

Lange, 2017-05-20:
Hello! :wave:

I'm the co-author of an npm module called [`elgato-stream-deck`](https://github.com/Lange/node-elgato-stream-deck) which provides an API for a specific hardware device. We're working on a [pull request](https://github.com/Lange/node-elgato-stream-deck/pull/27#discussion_r117613900) to add support for detecting when Stream Deck devices are plugged or unplugged, and `node-usb-detection` is a perfect fit for what we're trying to do.

However, compiling `node-usb-detection` requires a **lot** of dependencies on Windows, which means that installing `elgato-stream-deck` is difficult for users. Is there any possibility that we could assist in getting pre-built binaries set up for this module, to avoid end users needing to install a full compiler toolchain?

Thanks for this module! 👍

MadLittleMods, 2017-05-20:
Hey @Lange,

If we can automate the build in CI and push it back to the appropriate spot, that sounds great. Not really interested in having a manual setup or a one-off provided binaries that you build.

Lange, 2017-05-20:
Agreed, we'd want it to be an automated process as well. I must admit that I have no experience in this area, but I do have some experience in making automated builds of Electron apps on Travis and Appveyor and then publishing those artifacts to GitHub, so hopefully that knowledge will help here.

Lange, 2017-05-20:
I see that the documentation states that some of the tests require manual human intervention to plug and unplug a device. Would you want to simply skip these tests in CI?

ProbablePrime, 2017-05-20:
Some use of mocking in: https://github.com/MadLittleMods/node-usb-detection/blob/master/index.js#L64 could allow for Unit tests that can run on a CI.

- Mock the lower level part
- Make it emit events that look like devices connecting and disconnecting

The existing manual intervention tests are better described as integration tests IMO. they're still valid but Integration could be moved to not be on the CI

MadLittleMods, 2017-05-20:
@Lange I think we should only run the pre-built binary builds on tagged releases. Which would mean the release already went through some QA to get to that point and we would be getting to that point.

Lange, 2017-05-20:
@MadLittleMods Got it -- [electron-builder](https://github.com/electron-userland/electron-builder) does something similar.

jayalfredprufrock, 2017-08-12:
Would love to see this happen as well. I'm going to fork this and try my hand at using https://github.com/mafintosh/prebuild for now. Looks like it's pretty straightforward and even allows you to target electron.

jayalfredprufrock, 2017-08-12:
This shows how it works: https://github.com/iwinks/node-usb-detection
You include `prebuild` as a dev dependency and `prebuild-install` as a regular dependency. Then add a script for the build process and a script for the install and it automatically creates releases like so: https://github.com/iwinks/node-usb-detection/releases

In this case I don't think a pull request is helpful since this would need to get tied into your build system / CI server and it also requires setting up a Github token so prebuild can upload the binaries. All in all though, this process was incredibly easy and provides wonderful convenience for people just getting their feet wet with native node modules (which seem to be on the rise from projects like Electron).

todbot, 2017-08-14:
Thanks @jayalfredprufrock for pointing out `prebuild`!  I maintain `node-hid` and have been struggling under the weight of multi-version cross-platform issues, which I think `prebuild` will help with.  I see that `node-serialport` is also using `prebuild`.  It looks like a great tool.

timfish, 2017-09-25:
One thing worth noting is that binaries built on Ubuntu 16.04 fail to run on RHEL7 with the following error:
`/lib64/libstdc++.so.6: version 'GLIBCXX_3.4.21' not found`

For this reason, you'll want to ensure that any prebuilt binaries are build on Ubuntu 14.04 or something similarly old for compatability!

Lange, 2017-11-07:
@MadLittleMods just checking in on this, it looks like @jayalfredprufrock was able to lay out [a quick path forward in his comment](https://github.com/MadLittleMods/node-usb-detection/issues/40#issuecomment-321998442), which requires minimal code changes but _does_ require setting up some tokens and env vars on CI.

If you like (and if you're willing to trust a random person on GitHub), I'd be willing to do the legwork of getting this all set up if you want to grant me the appropriate permissions.

MadLittleMods, 2017-11-10:
Thanks for the PoC @jayalfredprufrock 💯 I just learned about how `node-v57` corresponds to the ABI (Application binary interface) version of Node (N-API), ex. mapping https://github.com/lgeiger/node-abi

 - Does it cover all platforms (Windows, macOS, and Linux)?
     - I see that there are a couple `win32-x64` builds for Windows and the rest are `darwin-x64` for macOS, https://github.com/iwinks/node-usb-detection/releases
     - Looking at https://github.com/node-serialport/node-serialport/releases/tag/v6.0.4, I see they have `win32-ia32`, `win32-x64`, `darwin-x64`, `linux-ia32`, `linux-x64` builds so I assume it is possible.
 - Can it all be built from a linux box?

@Lange Feel free to submit a PR with the setup from [@jayalfredprufrock fork](https://github.com/iwinks/node-usb-detection) and the `.travis-ci.yml`(or other preferred CI???). I can add the few necessary secrets, etc once the base is ready.

Lange, 2017-11-10:
@MadLittleMods macOS and Linux will likely need to be built on Travis, and be done as separate jobs (i.e. the Linux build is done on a Linux container, the macOS build is done on one of Travis' mac build workers). The Windows builds will likely need to be done via [AppVeyor](https://www.appveyor.com/), the predominant Windows-based CI service. However, there may be shortcuts I don't know about which @jayalfredprufrock could fill me in on.

As far as PRs go, it looks like all that needs to change are a few lines in `package.json`. The diff from Jay's fork illustrates this: https://github.com/MadLittleMods/node-usb-detection/compare/master...iwinks:master

I'm assuming that the majority of the legwork has been done on the UIs of the CI services themselves, as I don't see any YML files for CI configuration in Jay's fork.

MadLittleMods, 2017-11-10:
@Lange AFAICT, @jayalfredprufrock releases were just done manually locally. I think it would require a CI config to specify we only want that CI task to be run on tagged releases and which scripts to run. Most of my experience comes from GitLab CI (biased GitLab employee) but I have setup basic TravisCI configs.

Lange, 2017-11-10:
@MadLittleMods Ahh okay. I have experience setting up complex CI configs for Travis, Circle, and AppVeyor. I'll fork this repo and take a crack at setting up Linux, macOS, and Windows builds, and report back with my findings.

Lange, 2017-11-10:
Status update: making great progress and I expect to have a PR open within 24 hours. Fun side note: I found a bug in the build scripts for `node-serialport` while studying them: https://github.com/node-serialport/node-serialport/issues/1397

closed 2017-11-11 by MadLittleMods