MadLittleMods/node-usb-detection #41: Rebuild node v8.x.x error

ninox92, 2017-06-15:
I'm trying to rebuild node-usb-detection for ARM on node 8.1.2.
When rebuilding, i receive the following error.
Is this a problem with node-gyp or usb-detection?

where does command line option '-m64' come from ?

```
 > usb-detection@1.4.0 install /home/thomas/git/fys/build/tmp-glibc/work/cortexa9hf-neon-oe-linux-gnueabi/fysiologic-cradle-software/1.0-r0/git/node_modules/usb-detection
| > node-gyp rebuild
| 
| make: Entering directory `/home/thomas/git/fys/build/tmp-glibc/work/cortexa9hf-neon-oe-linux-gnueabi/fysiologic-cradle-software/1.0-r0/git/node_modules/usb-detection/build'
|   CXX(target) Release/obj.target/detection/src/detection.o
| arm-oe-linux-gnueabi-g++: error: unrecognized command line option '-m64'
| make: *** [Release/obj.target/detection/src/detection.o] Error 1
| make: Leaving directory `/home/thomas/git/fys/build/tmp-glibc/work/cortexa9hf-neon-oe-linux-gnueabi/fysiologic-cradle-software/1.0-r0/git/node_modules/usb-detection/build'
| gyp ERR! build error
| gyp ERR! stack Error: `make` failed with exit code: 2
| gyp ERR! stack     at ChildProcess.onExit (/home/thomas/git/fys/build/tmp-glibc/sysroots/x86_64-linux/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
| gyp ERR! stack     at emitTwo (events.js:125:13)
| gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
| gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:197:12)
| gyp ERR! System Linux 4.4.0-79-generic
| gyp ERR! command "/home/thomas/git/fys/build/tmp-glibc/sysroots/x86_64-linux/usr/bin/node" "/home/thomas/git/fys/build/tmp-glibc/sysroots/x86_64-linux/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
| gyp ERR! cwd /home/thomas/git/fys/build/tmp-glibc/work/cortexa9hf-neon-oe-linux-gnueabi/fysiologic-cradle-software/1.0-r0/git/node_modules/usb-detection
| gyp ERR! node -v v8.1.2
| gyp ERR! node-gyp -v v3.6.2
| gyp ERR! not ok
| npm ERR! code ELIFECYCLE
| npm ERR! errno 1
| npm ERR! usb-detection@1.4.0 install: `node-gyp rebuild`
| npm ERR! Exit status 1
| npm ERR!
| npm ERR! Failed at the usb-detection@1.4.0 install script.

```

MadLittleMods, 2017-06-15:
It builds and the tests pass for me on Windows 10 with `node@8.1.2`

ninox92, 2017-06-16:
Fixed!
I cleaned up to much files. So node-gyp was searching for a file that did not exist.
```
env V=1 node-gyp rebuild
```
The command above helped me to find the solution

closed 2017-06-16 by MadLittleMods