MadLittleMods/node-usb-detection #44: Access Violation on Window 10

timfish, 2017-09-25:
Thanks for this module!

In electron 1.7.8 I've recorded this access violation but have no idea how to reproduce it:

```
NotifyRemoved(ListResultItem_t *)
  detection.cpp:87
RtlpUnWaitCriticalSection
  ntdll.dll
```
I've got the full dump file if that'll help.

[image]

MadLittleMods, 2017-10-02:
@timfish It seems like you are past bundling `usb-detection` in your electron app but did you make sure to rebuild the package with electron, https://github.com/MadLittleMods/node-usb-detection/issues/39#issuecomment-315524803

I don't have any experience using this in Electron. If you can come up with some reproducible steps, I can check it out.

timfish, 2017-10-03:
Yes, building and bundling is all setup. We use electron-builder which makes this all work with very little effort. The library is working well on Windows/Mac and Linux apart from this crash report I received. It can probably be ignored unless we start seeing a lot of occurrences!

closed 2017-10-03 by timfish

wujiazhong, 2017-12-16:
I met this crash issue that cannot access virtual address when building and bundling electron app with electron-packager on Windows 7 & 10. It works well on mac.
[image]