MadLittleMods/node-usb-detection #45: add and remove event get different device object

hungtcs, 2017-11-09:
Hello! This is my screenshot:

[image]

MadLittleMods, 2017-11-09:
Hey @hungtcs,

Which OS are you on?

hungtcs, 2017-11-10:
Hello @MadLittleMods, I use elementary OS 0.4.1 Loki, based on Ubuntu 16.04.3

MadLittleMods, 2017-11-10:
Are you able to test on another OS with that same device?

I just tested on a Ubuntu 16.04.3 vm with the latest `master` (almost the same as latest release `1.4.0`) with a microphone and I see the same device object.

Add
```
{
  locationId: 0,
  vendorId: 2313,
  productId: 24,
  deviceName: 'AT2020_USB',
  manufacturer: 'audio-technica_Ƞ',
  serialNumber: '',
  deviceAddress: 0
}
```

Remove
```
{
  locationId: 0,
  vendorId: 2313,
  productId: 24,
  deviceName: 'AT2020_USB',
  manufacturer: 'audio-technica_Ƞ',
  serialNumber: '',
  deviceAddress: 0
}
```