MadLittleMods/node-usb-detection #52: Module Version Mismatch

mm739521, 2017-12-07:
I have USB-Detection module installed in my application and just recently upgraded my electron version to accomodate another module.  When I run now, I'm getting the following stack trace error:

 App threw an error during load
 Error: Module
  version mismatch. Expected 50, got 54.
     at Error (native)
     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
     at Object.Module._extensions..node (module.js:583:18)
     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
     at Module.load (module.js:473:32)
     at tryModuleLoad (module.js:432:12)
     at Function.Module._load (module.js:424:3)
     at Module.require (module.js:483:17)
     at require (internal/module.js:20:19)
     at bindings (C:\pc2k_dev\device-sync\node_modules\usb-detection\node_modules\bindings\bindings.js:76:44)
     at Object.<anonymous> (C:\pc2k
 _dev\device-sync\node_modules\usb-detection\index.js:10:37)
     at Module._compile (module.js:556:32)
     at Object.Module._extensions..js (module.js:565:10)
     at Module.load (module.js:473:32)
     at tryModuleLoad (module.js:432:12)
     at Function.Module._load (module.js:424:3)
     at Module.require (module.js:483:17)
     at require (internal/module.js:20:19)
     at Object.<anonymous> (C:\pc2k_dev\device-sync\main.js:178:17)
     at Module._compile (module.js:556:32)
     at Object.Module._extensions..js (module.js:565:10)
     at Module.load (module.js:473:32)
 in process, on uncaughtException, err: {}
 Error: Module version mismatch. Expected 50, got 54.
     at Error (native)
     at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
     at Object.Module._extensions..node (module.js:583:18)
     at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
     at Module.load (module.js:473:32)
     at tryModuleLoad (module.js:432:12)
     at Function.Module._load (module.js:424:3)
     at Module.require (module.js:483:17)
     at require (internal/module.js:20:19)
     at bindings (C:\pc2k_dev\device-sync\node_modules\usb-detection\node_modules\bindings\bindings.js:76:44)
     at Object.<anonymous> (C:\pc2k_dev\device-sync\node_modules\usb-detection\index.js:10:37)
     at Module._compile (module.js:556:32)
     at Object.Module._extensions..js (module.js:565:10)
     at Module.load (module.js:473:32)
     at tryModuleLoad (module.js:432:12)
     at Function.Module._load (module.js:424:3)
     at Module.require (module.js:483:17)
     at require (internal/module.js:20:19)
     at Object.<anonymous> (C:\pc2k_dev\device-sync\main.js:178:17)
     at Module._compile (module.js:556:32)
     at Object.Module._extensions..js (module.js:565:10)
     at Module.load (module.js:473:32)

Any idea how to fix this?

MadLittleMods, 2017-12-07:
@mm739521 My guess is that you need to build with electron, see https://github.com/MadLittleMods/node-usb-detection/issues/39#issuecomment-315524803

todbot, 2017-12-07:
Yup, sounds like you need to do `electron-rebuild`

closed 2017-12-08 by MadLittleMods