MadLittleMods/node-usb-detection #53: Events no longer fired after stopping and restarting monitoring

timfish, 2018-01-19:
We're using this library rather successfully in our Electron app but I'm trying to fix an issue on both Windows and macOS.

We use Electrons [powerMonitor API](https://electronjs.org/docs/api/power-monitor) to `stopMonitoring` and `startMonitoring` from the `suspend` and `resume` events respectively. After stopping and restarting monitoring, we no longer get USB change events. Is this the expected behaviour?

timfish, 2018-01-19:
Ok, so my initial thoughts were why are we even disabling usb detection when the device is suspended and resumed. Testing on Windows 10 and macOS High Sierra seems to suggest that usb detection works perfectly well after coming out of suspend even if monitoring was left enabled throughout sleep. 

This then begs the question whether it's possible to restart monitoring after its been stopped. If the `stopMonitoring` function is only for cleanup before exiting the process, it might be cleaner to use the node.js [AtExit hooks](https://nodejs.org/api/addons.html#addons_atexit_hooks) and then consumers of this library don't have to worry about it anymore?

todbot, 2018-01-19:
Hi @timfish, 
Using AtExit would be the smart thing to do, but it looks like Electron 1.8 breaks that functionality currently: https://github.com/electron/electron/issues/11299

timfish, 2018-01-20:
I actually only found out that AtExit exists from that bug 🙂

MadLittleMods, 2018-02-04:
@timfish Sorry for the delay. In the 2.x release, I tried removing the side-effects and require `startMonitoring` to be called before things kick off but I didn't sufficiently test things and there are still side-effects.

I need to get https://github.com/MadLittleMods/node-usb-detection/pull/46 fixed up and merged and also look into ensuring start/stop work regardless of how many times called (help welcome).

`stopMonitoring` was more meant to be called once in order to remove all the blocking threads so your app can exit gracefully.

On macOS, I suspect after stopping, you can't start again because we stop those threads and don't start them back up in `startMonitoring`, see https://github.com/MadLittleMods/node-usb-detection/blob/9164178f40c28b0d5e2407be42f5aca9c8768e78/src/detection_mac.cpp#L390

On Windows, `startMonitoring` may work after `stopMonitoring` but I haven't tested it yet, https://github.com/MadLittleMods/node-usb-detection/blob/9164178f40c28b0d5e2407be42f5aca9c8768e78/src/detection_win.cpp#L167