MadLittleMods/node-usb-detection #55: cross-platform support

ddombrowsky, 2018-02-19:
This is more of a question than an issue:  How best to deploy this `usb-detection` module in a platform independent way?  Since the `cpp` files are platform specific, and the `detection.node` file is the same, I need to have a copy of this module directory for every platform my app runs on.  

Is there a better way to do this?  Perhaps the `require('bindings')` line can take the platform and attempt to load a differently named bindings file?  I'll probably end up doing it this way, but I wanted to ask first to make sure this wasn't a solved problem.

ddombrowsky, 2018-02-19:
In order for that to work, wouldn't every user that uses my app need to download something from the Internet when running the app for the first time?  That breaks my "no internet required to run" requirement.

todbot, 2018-02-19:
What is your app? A desktop app via Electron?  You'll need to create separate binaries for each OS you plan to support.

ddombrowsky, 2018-02-19:
Correct. It is an electron-based app.  Right now I manually rename `node_modules/usb-detection` -> `node_modules/usb-detection.linux` and to a `require('usb-detection.linux')` (or win32) depending on which platform I'm running on.  It seems like that is something that _could_ be handled within the `usb-detection` project itself.  I _think_ all that is required is to have a different version of the `detection.node` file loaded based on the platform the code is running on.  Is that right?

todbot, 2018-02-19:
You should be using `electron-builder` or `electron-forge`. Both take an Electron-based project and create full executables for each platform you care about.  So you don't have to do this yourself.

Lange, 2018-02-19:
Here's a pretty simple example of an electron app being built with `electron-builder`, if you'd like another example to work against: https://github.com/Lange/x32-mute-matrix

It doesn't use `node-usb-detection`, but the concepts are the same regardless.

MadLittleMods, 2018-02-19:
Thanks @todbot and @Lange 💯 

@ddombrowsky Judging from the :+1:, I assume this sent you in the right direction. Feel free to re-open if you need some more help.

closed 2018-02-19 by MadLittleMods

ddombrowsky, 2018-02-20:
Yes, you can leave this closed.  I'm not sure if `electron-forge` or `electron-builder` are going to work, but I'm also convinced that this is out of scope as an issue for this project.  All solutions basically assume the user is going to say "Hey, I'm windows!  I'll download the windows installer!"  But I can't do that.  The same code needs to run from the same USB stick, without Internet access, on the three main platforms.

lishichao1002, 2018-09-13:
> Here's a pretty simple example of an electron app being built with `electron-builder`, if you'd like another example to work against: https://github.com/Lange/x32-mute-matrix
> 
> It doesn't use `node-usb-detection`, but the concepts are the same regardless.

in your project, 
//step1
modify package.json add ia32
```
"win": {
      "asar": false,
      "target": [
        {
          "target": "nsis",
          "arch": [
            "ia32"
          ]
        }
      ]
    }
```
//step2
modify app/package.json add usb-detection
```
{
  "name": "x32-mute-matrix",
  "version": "1.0.1",
  "description": "See and configure the mute status of all your X32's channels (on every mixbus) in one place.",
  "homepage": "https://github.com/lange/x32-mute-matrix",
  "dependencies": {
    "electron-debug": "^1.0.0",
    "electron-is-dev": "^0.3.0",
    "electron-log": "^2.2.7",
    "electron-updater": "^2.21.8",
    "electron-window-state": "^4.1.1",
    "get-port": "^3.2.0",
    "osc": "^2.2.0",
    "tinycolor2": "^1.4.1",
    "usb-detection": "^3.2.0"
  },
  "repository": {
    "type": "git",
    "url": "https://github.com/lange/x32-mute-matrix.git"
  },
  "license": "MIT",
  "author": "Alex Van Camp <email@alexvan.camp>",
  "main": "server/main.js"
}
```
// step3: import usb-detection
```
//app/server/main.js
// Native
const path = require('path');

// Packages
const windowStateKeeper = require('electron-window-state');
const {app, BrowserWindow} = require('electron');
const log = require('electron-log');

// Ours
const {version} = require('./util');

const usbDetection = require('usb-detection');
console.log(usbDetection);
```

//step4: npm run build , error 
```
PS Z:\Users\lishichao\workspaces\webstorm\x32-mute-matrix> npm run dist

[... 90 lines cut ...]
From previous event:
    at Object.args [as handler] (Z:\Users\lishichao\workspaces\webstorm\x32-mute-matrix\node_modules\electron-builder\out\cli\cli.js:184:86)
    at Object.runCommand (Z:\Users\lishichao\workspaces\webstorm\x32-mute-matrix\node_modules\electron-builder\node_modules\yargs\lib\command.js:235:44)
    at Object.parseArgs [as _parseArgs] (Z:\Users\lishichao\workspaces\webstorm\x32-mute-matrix\node_modules\electron-builder\node_modules\yargs\yargs.js:1042:24)
    at Object.get [as argv] (Z:\Users\lishichao\workspaces\webstorm\x32-mute-matrix\node_modules\electron-builder\node_modules\yargs\yargs.js:957:21)
    at Object.<anonymous> (Z:\Users\lishichao\workspaces\webstorm\x32-mute-matrix\node_modules\electron-builder\out\cli\cli.js:179:373)
    at Module._compile (module.js:569:30)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:503:32)
    at tryModuleLoad (module.js:466:12)
    at Function.Module._load (module.js:458:3)
    at Function.Module.runMain (module.js:605:10)
    at startup (bootstrap_node.js:158:16)
    at bootstrap_node.js:575:3
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! x32-mute-matrix@1.0.1 dist: `electron-builder`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the x32-mute-matrix@1.0.1 dist script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\lishichao\AppData\Roaming\npm-cache\_logs\2018-09-13T09_37_24_759Z-debug.log
```