MadLittleMods/node-usb-detection #56: node-usb-detection in electron for barcode scanner

lalitmee, 2018-02-27:
Hello,
I am trying to use node-usb-detection in my electron desktop application for a barcode scanner. I am using it will following details:- 
- OS :- Ubuntu 16.04
- node version :- v9.3.0
- npm version :- 5.6.0
- electron version :- v1.8.2
- webpack version :- 3.11.0
- node-usb-detection version :- 1.0.2
- React.js version :- 0.14.8

The problem which I am facing is that when I am running webpack, it is showing two warnings which are following :- 
```
WARNING in ./node_modules/bindings/bindings.js
76:22-40 Critical dependency: the request of a dependency is an expression
 @ ./node_modules/bindings/bindings.js
 @ ./node_modules/usb-detection/index.js
 @ ./app/app.js

WARNING in ./node_modules/bindings/bindings.js
76:43-53 Critical dependency: the request of a dependency is an expression
 @ ./node_modules/bindings/bindings.js
 @ ./node_modules/usb-detection/index.js
 @ ./app/app.js
```
And when I am running electron application, nothings is showing. On inspecting, there is one error in console which is as follows :- 
``` 
Uncaught Error: Could not locate the bindings file. Tried:
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/build/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/build/Debug/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/build/Release/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/out/Debug/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/Debug/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/out/Release/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/Release/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/build/default/detection.node
 → /home/lalit/Desktop/Github/Electron/simple-electron/simple-electron-react/compiled/5.1.1/linux/x64/detection.node
```

For reproducing the error, I am mentioning my [repo](https://github.com/lalitmee/simple-electron-react) for this project. 
- Steps to reproduce :- 
```
git clone https://github.com/lalitmee/simple-electron-react.git
cd simple-electron-react
npm install
npm run electron (for webpack and here it will show two warnings)
npm start
```
I am trying to solve this from a long time but I am unable to solve. Please help me. It is very important part of my application.
Thank you so much.

todbot, 2018-02-27:
I've not tried this yet (but want to), but try using
 https://github.com/webpack-contrib/node-loader

lalitmee, 2018-02-27:
@todbot okk, thanks for the suggestion. I will try to use this and will report the result here. :innocent:

todbot, 2018-02-27:
actually perhaps nevermind, from this issue: https://github.com/webpack/webpack/issues/5762
I think the preferred solution is to put the native modules in the `"externals"` section of webpack.config.js.  This is what I do in Blink1Control2 for `node-hid`, but I admit I know very little about webpack: https://github.com/todbot/Blink1Control2/blob/master/webpack.config.js

lalitmee, 2018-02-27:
oh.. I saw your **webpack.config.js** file and I think the step which you are telling can help me. But as you are using in your externals 
```
  externals: {
    "express": 'commonjs express',
    "node-hid": 'commonjs node-hid',
    'nconf': 'commonjs nconf',
    'xml2js': 'commonjs xml2js' // this is to keep webpack from complaing about optional dep of 'needle'
  },
```
So what should I use for **node-usb-detection**. I am also new to webpack that's why.
```
externals: {
    "node-usb-detection" : '?'
  },
```
What should be there in place of **?**. Thanking you. :smile:

todbot, 2018-02-27:
I think it would look like:
```
externals: {
    "node-usb-detection" : 'commonjs node-usb-detection'
},
```

MadLittleMods, 2018-05-10:
Closing as I assume you figured it out (no response). @lalitmee If you found a solution, please post it here so others can benefit.

closed 2018-05-10 by MadLittleMods