MadLittleMods/node-usb-detection #57: Calling startMonitoring() and stopMonitoring() causes segmentation fault on Ubuntu 16.04

qtphan734, 2018-04-03:
In a simple node application, if you execute the folllowing code, you will either get segmentation fault or some memory corruption error on Ubuntu 16.04:

usbDetection.startMonitoring();
usbDetection.stopMonitoring();
usbDetection.startMonitoring();
usbDetection.stopMonitoring();

MadLittleMods, 2018-04-03:
Related to https://github.com/MadLittleMods/node-usb-detection/issues/53

PeterBurner, 2019-09-16:
I don't think calling them multiple times ist the problem. `stopMonitoring()` seems to be broken on linux.
I can generate a segmentation fault with the following code:
```javascript
var usbDetect = require('usb-detection');

usbDetect.startMonitoring();

usbDetect.stopMonitoring();

setTimeout(() => console.log('end'), 1000);
```
If I comment out the line with `stopMonitoring()` I get a log output but afterwards the process hangs.

Node Version: v12.8.0
OS: Raspbian 10 (buster)

By the way: The code above works just fine on MacOS.

P0oOOOo0YA, 2019-12-31:
I can reproduce the problem on my Linux machine with the code posted by @PeterBurner . 

```sh
    OS: Linux 5.3 Kali GNU/Linux Rolling 2019.4
    Node: 13.5.0
```

gre, 2020-01-02:
I reproduce this situation in usb-detection@4.7.0 on Node 12 and Mac.

frankrobert-ls, 2020-02-13:
As @gre mentioned this can be reproduced on Mac with Node12. Could we update this issue's title + labels to reflect it better? 🌈

kfazz, 2021-07-09:
For 4.10, I was able to fix this with the following patch:

--- node-usb-detection-4.10.0.orig/src/detection_linux.cpp
+++ node-usb-detection-4.10.0/src/detection_linux.cpp
@@ -98,6 +98,7 @@ void Stop() {
 	uv_cond_destroy(&notifyDeviceHandled);
 
 	udev_monitor_unref(mon);
+	mon = NULL;
 	udev_unref(udev);
 }
 
@@ -226,19 +227,21 @@ static void cbWork(uv_work_t *req) {
 		if (!ret) continue;
 		if (ret < 0) break;
 
-		dev = udev_monitor_receive_device(mon);
-		if (dev) {
-			if(udev_device_get_devtype(dev) && strcmp(udev_device_get_devtype(dev), DEVICE_TYPE_DEVICE) == 0) {
-				if(strcmp(udev_device_get_action(dev), DEVICE_ACTION_ADDED) == 0) {
-					WaitForDeviceHandled();
-					DeviceAdded(dev);
-				}
-				else if(strcmp(udev_device_get_action(dev), DEVICE_ACTION_REMOVED) == 0) {
-					WaitForDeviceHandled();
-					DeviceRemoved(dev);
+		if(mon) {
+			dev = udev_monitor_receive_device(mon);
+			if (dev) {
+				if(udev_device_get_devtype(dev) && strcmp(udev_device_get_devtype(dev), DEVICE_TYPE_DEVICE) == 0) {
+					if(strcmp(udev_device_get_action(dev), DEVICE_ACTION_ADDED) == 0) {
+						WaitForDeviceHandled();
+						DeviceAdded(dev);
+					}
+					else if(strcmp(udev_device_get_action(dev), DEVICE_ACTION_REMOVED) == 0) {
+						WaitForDeviceHandled();
+						DeviceRemoved(dev);
+					}
 				}
+				udev_device_unref(dev);
 			}
-			udev_device_unref(dev);
 		}
 	}
 }

closed 2022-03-02 by MadLittleMods