MadLittleMods/node-usb-detection #58: Can't use module in 32-bit applications

TacheRadu, 2018-04-14:
The name says it all: I don't know how to make it work, testing it is okay, because I have the 64-bit version of node and npm installed, but when I build for 32-bit, it doesn't work, giving me the "Error: %1 is not a valid Win32 application."

doganmurat, 2018-04-14:
Are you compiling for electron?

TacheRadu, 2018-04-14:
I made it work in the end, here's what I did:
`npm install --save --arch=ia32 usb-detection`
then I made a script inside the package.json: `"rebuild": "electron-rebuild --arch=ia32"`
and I ran it: `npm run rebuild`
After that, it worked like it was supposed to, hope this helps anyone having the same issue.

closed 2018-04-14 by TacheRadu

mm739521, 2018-04-23:
Are you using electron-builder for your builds?  I'm having the same `Error: %1 is not a valid Win32 Application`.  I ran all the commands you did then do my build and it didn't fix it for me.

TacheRadu, 2018-04-23:
I've actually been using electron-packager, and the command I've used is `electron-packager . Electron --platform win32 --arch ia32 --asar --out dist/`
I hope it helps you.

todbot, 2018-04-23:
I think with `electron-packager` you still need to run `electron-rebuild` beforehand to get the native library part of `node-usb-detection` compiled for your OS + Electron version.

lishichao1002, 2018-09-13:
+1

```
PS Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron> npm install --save --arch=ia32 usb-detection

> usb-detection@3.2.0 install Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection
> prebuild-install || node-gyp rebuild

prebuild-install WARN install %1 is not a valid Win32 application.
prebuild-install WARN install \\?\Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection\build\Release\detection.node

Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node ""
rebuild )
在此解决方案中一次生成一个项目。若要启用并行生成，请添加“/m”开关。
  detection.cpp
  deviceList.cpp
  detection_win.cpp
..\src\detection.cpp(60): warning C4996: 'Nan::Callback::Call': was declared deprecated [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection\build\detection.vcxproj]
  Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\nan\nan.h(1647): note: see declaration of 'Nan::Callback::Call'
..\src\detection.cpp(105): warning C4996: 'Nan::Callback::Call': was declared deprecated [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection\build\detection.vcxproj]
  Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\nan\nan.h(1647): note: see declaration of 'Nan::Callback::Call'
..\src\detection.cpp(195): warning C4996: 'Nan::Callback::Call': was declared deprecated [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection\build\detection.vcxproj]
  Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\nan\nan.h(1647): note: see declaration of 'Nan::Callback::Call'
  win_delay_load_hook.cc
..\src\detection_win.cpp(5): fatal error C1083: Cannot open include file: 'atlstr.h': No such file or directory [Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection\build\detection.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:125:13)
gyp ERR! stack     at ChildProcess.emit (events.js:213:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 6.1.7601
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd Z:\Users\lishichao\workspaces\webstorm\soda-b-pc\dist\electron\node_modules\usb-detection
gyp ERR! node -v v8.2.1
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok
npm WARN com.didichu.bpc.app@0.1.0 No repository field.
npm WARN com.didichu.bpc.app@0.1.0 No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb-detection@3.2.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the usb-detection@3.2.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\lishichao\AppData\Roaming\npm-cache\_logs\2018-09-13T06_45_22_061Z-debug.log
```

mikew, 2019-04-06:
Sucks we need to install VS Proper just for this library. Apparently Windows Driver Kit 7.1.0 contains the needed headers, but I can't seem to get node-gyp / this project to use them. They're installed to `C:\WinDDK\7600.16385.1\inc\atl71`.

MadLittleMods, 2019-04-08:
@mikew If you just `npm install usb-detection` it should try to use one of the prebuilt binaries, https://github.com/MadLittleMods/node-usb-detection/releases

mikew, 2019-04-08:
I should have clarified. This was with Electron. Fix was found with `windows-build-tools` here https://github.com/felixrieseberg/windows-build-tools/issues/92#issuecomment-480524316.