MadLittleMods/node-usb-detection #63: [Question] IOCreatePlugInInterfaceForService Stderr Message

thewalrusisben, 2018-05-31:
Hey, thanks for making this package! This is a big help for a project I'm working on. 

One question though: I'm noticing that when the package initializes on macOS (I'm running 10.13.4 currently), it starts of by logging "IOCreatePlugInInterfaceForService returned 0xe00002be" multiple times. I tracked this down and it appears to be because it's trying to create a plugin interface for the AppleUSBXHCI Root Hub Simulation and iBridge.

I'm assuming the reason for this is because these are virtual devices that don't necessarily have their own USB interfaces. Is there a reason to be concerned about these messages (this will be running in a production environment), and do you think there's any reason not to just filter these kinds of devices out entirely on macOS?

Thanks!

MadLittleMods, 2018-05-31:
Only listing devices with an interface was discussed a bit here, https://github.com/MadLittleMods/node-usb-detection/pull/50#issuecomment-390489242

Is it working fine outside of the messages?

It looks like our code is meant to gracefully continue if that happens, https://github.com/MadLittleMods/node-usb-detection/blob/61dd19d517a8ef757457e0b6d7c2095367f28d3b/src/detection_mac.cpp#L223

thewalrusisben, 2018-06-15:
After a couple weeks of running on ~ 20 devices in production, this definitely hasn't caused any issues which, as you point out, definitely would be expected. It'd be aesthetically nice to not have the log message in there from my perspective, but this is by no means a problem! Thanks!

closed 2018-06-15 by thewalrusisben