MadLittleMods/node-usb-detection #64: Certain USB devices are not detected

AaronBeaudoin, 2018-07-16:
Everything works fine when I connect a generic mouse or keyboard, but no event is fired when I connect a particular bar code scanner I need my application to respond to. This is strange because not only does the scanner show up without issue in Device Manager (I'm on Windows 10), but I can also actually connect to the scanner without any issue using the Node.js package `node-hid`. The scanner shows up in the list of items when I run `HID.devices()`. I know that `node-hid` is a separate package, but I just wanted to ensure that the issue was not with the scanner or something else.

Here are the details that `node-hid` gives me for the scanner:
```
{
  "vendorId": 1504,
  "productId": 1536,
  "path": "\\\\?\\hid#vid_05e0&pid_0600#6&30439c8b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
  "serialNumber": "S\/N:B5CA45CDF65E Rev:NBRMKAAB5",
  "manufacturer": "\ufffdSymbol Technologies, Inc, 2002",
  "product": "Symbol Bar Code Scanner",
  "release": 513,
  "interface": -1,
  "usagePage": 65283,
  "usage": 3072
}
```

Any idea why `usb-detection` isn't doing anything when I connect or disconnect it?

MadLittleMods, 2018-07-16:
@AaronBeaudoin I don't know why it wouldn't be detected

 - Please share the code you are trying to use
 - What version of `usb-detection` are you using?
 - What OS are you on?
 - Do other USB devices get detected with `usb-detection`?
 - Is it listed when you do a `usbDetect.find(...)`?
 - Do you see an event when inserted `usbDetect.on('add', function(device) { ... })`?

AaronBeaudoin, 2018-07-17:
Here is the code I've been testing with:
```
const usbDetect = require("usb-detection")
usbDetect.startMonitoring()

let logDevices = (msg, device) => {
  console.log(msg, device.vendorId, device.productId, device.deviceName)
  usbDetect.find((err, devices) => {
    devices.map((device) => console.log(
      "find", device.vendorId, device.productId, device.deviceName))
  })
}

usbDetect.on("add", (device) => logDevices("add", device))
usbDetect.on("remove", (device) => logDevices("remove", device))
```

I am using `usb-detection` 3.1.0.
My OS is Windows 10 Pro Version 1803 Build 17134.1.
Here is some output from the code above (with comments):
```
// Mouse and bar code scanner disconnected initially
// I connect the mouse

add 1133 49271 USB Input Device
find 1121 16 USB Input Device
find 1121 16 USB Input Device
find 32903 32776 Generic USB Hub
find 32903 32768 Generic USB Hub
find 1121 16 USB Composite Device
find 1121 19727 USB Input Device
find 1133 49271 USB Input Device
find 0 0 USB Root Hub
find 0 0 USB Root Hub
find 0 0 USB Root Hub (USB 3.0)

// As you can see, the mouse is in the list

// I connect the bar code scanner... nothing
// I disconnect the mouse

remove 1133 49271 USB Input Device
find 1121 16 USB Input Device
find 1121 16 USB Input Device
find 32903 32776 Generic USB Hub
find 32903 32768 Generic USB Hub
find 1121 16 USB Composite Device
find 1121 19727 USB Input Device
find 0 0 USB Root Hub
find 0 0 USB Root Hub
find 0 0 USB Root Hub (USB 3.0)

// Even though the bar code scanner was still connected
// it didn't show up in the list

// I end the script and start it again
// I plug in the mouse again

add 1133 49271 USB Input Device
find 1121 16 USB Input Device
find 1121 16 USB Input Device
find 32903 32776 Generic USB Hub
find 32903 32768 Generic USB Hub
find 1121 16 USB Composite Device
find 1121 19727 USB Input Device
find 1133 49271 USB Input Device
find 1504 1536 USB Input Device
find 0 0 USB Root Hub
find 0 0 USB Root Hub
find 0 0 USB Root Hub (USB 3.0)

// The bar code scanner is now in the list
// with a VID of 1504 and a PID of 1536
// as expected from the information in my last post
```

MadLittleMods, 2018-07-18:
Nothing stands out to me in your code.

The one thing that stands out to me about the device is `"interface": -1,`. Looking at all of my devices, they have `0`, `1`, `2`, etc

You can try some logging inside `DetectCallback` to see if the change is seen at all, https://github.com/MadLittleMods/node-usb-detection/blob/27a3e03831c63d31bd0149b6f627fecede451618/src/detection_win.cpp#L304

LanetheGreat, 2018-11-09:
I think I figured out what's happening. I'm having the same issue while developing an app for my workplace who also uses several Symbol handheld barcode scanners. I found out that `ListResultItem_t* currentDevice` is still NULL when `NotifyAdded`/`NotifyRemoved` is called because in `UpdateDevice` (in `detection_win.cpp`) when the for loop enumerates every device, the if statement: `if(szDevId == buf)` never matches up with our devices.

I modified the library to print out the `szDevId` (which is passed down from the WM_DEVICECHANGE message's lParam parameter aka the PDEV_BROADCAST_DEVICEINTERFACE structure) and `buf` from `UpdateDevice` and found that when there's a forward slash in the Device's id it's converted to a backslash when sent from WM_DEVICECHANGE vs a call to SetupDiGetDeviceInstanceId where it's left as is.

Here's the output where you'll notice the device that's suppose to match, doesn't because of the different formats given by the Windows API's:
```
~Enumerating devices.~
Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_0BDA&PID_0153\20120926571200000

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_8087&PID_800A\5&2F4BF314&0&1

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_03F0&PID_0024\5&76963A6&0&8

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_8087&PID_8002\5&14511B4E&0&1

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_04F2&PID_1459\5&76963A6&0&9

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_8087&PID_0A2A\5&76963A6&0&1

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\ROOT_HUB20\4&25F5B200&0

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\ROOT_HUB20\4&1AB1954D&0

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\VID_05E0&PID_1300\S/N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1

Device: USB\VID_05E0&PID_1300\S\N:3108642D88FB3B45A9F09F3CAA4FBB08_REV:CBRMIASC1
Buffer: USB\ROOT_HUB30\4&3AE6247&0&0
```

When I do a character find/replace on the `buf` string in `UpdateDevice` here in the code and `electron-rebuild`, it fixes it for me.
```c++
for(int i=0; DllSetupDiEnumDeviceInfo(hDevInfo, i, pspDevInfoData); i++) {
	DWORD nSize=0 ;
	TCHAR buf[MAX_PATH];

	if (!DllSetupDiGetDeviceInstanceId(hDevInfo, pspDevInfoData, buf, sizeof(buf), &nSize)) {
		break;
	}
	// Backslash fix.
	for(int j=0; j<MAX_PATH; j++)
		if(buf[j] == '/')
			buf[j] = '\\';
	// End fix.

	if(szDevId == buf) {
		WaitForSingleObject(deviceChangedSentEvent, INFINITE);

		DWORD DataT;
		DWORD nSize;
		DllSetupDiGetDeviceRegistryProperty(hDevInfo, pspDevInfoData, SPDRP_LOCATION_INFORMATION, &DataT, (PBYTE)buf, MAX_PATH, &nSize);
 		        DllSetupDiGetDeviceRegistryProperty(hDevInfo, pspDevInfoData, SPDRP_HARDWAREID, &DataT, (PBYTE)(buf + nSize - 1), MAX_PATH - nSize, &nSize);
```

MadLittleMods, 2018-11-11:
@LanetheGreat Mind submitting a PR? Perhaps we should convert all the forward slashes to backslash as that seems more common across the API on Windows

LanetheGreat, 2018-11-14:
#73 Created a pull request for the fix, tried to keep it in the same styling as your current code.

closed 2018-11-24 by MadLittleMods