MadLittleMods/node-usb-detection #68: How to get USB Intance ID?

hifong45, 2018-09-10:
I found instance-specific-ID in detection_win.cpp, but can't get from device["serialNumber"], it's "".

How to get it?

Thanks !

MadLittleMods, 2018-09-10:
@hifong45 Not all devices have a serial number. See for example here, https://github.com/MadLittleMods/node-usb-detection/pull/62#pullrequestreview-123573550

hifong45, 2018-09-11:
Excuse me, I want to get <instance-specific-ID>`: `5&376ABA2D&0&21’ in fact.
How can I do ?  I cant find the example or description
Thank you !!!!!

hifong45, 2018-09-11:
Excuse me, I want to get instance-specific-ID : 5&376ABA2D&0&21’ in fact.
How can I do ? I cant find the example or description
Thank you !!!!!

MadLittleMods, 2018-09-11:
@hifong45 If it's not appearing in the `serialNumber` property, then it might not be possible to get that information with this library yet.

You may have to correlate it with a device list from https://www.npmjs.com/package/usb or some other package.

hifong45, 2018-09-12:
Ok, thank you!

closed 2018-09-12 by MadLittleMods