MadLittleMods/node-usb-detection #71: how to use with angular 2

halilpolathalil, 2018-10-01:
Hi,
I want webusb detection but require not work form me. What should I use instead of require in angular 2

MadLittleMods, 2018-10-01:
Related to https://github.com/MadLittleMods/node-usb-detection/issues/67

> As a note, you can't use usb-detection on the frontend in a browser
>
> https://github.com/MadLittleMods/node-usb-detection/issues/67#issuecomment-407427174

What isn't working about it currently?

halilpolathalil, 2018-10-01:
> Related to #67
> 
> > As a note, you can't use usb-detection on the frontend in a browser
> > [#67 (comment)](https://github.com/MadLittleMods/node-usb-detection/issues/67#issuecomment-407427174)
> 
> What isn't working about it currently?

I can't use require() in that line var usbDetect = require('usb-detection');  Because angular 2 not support ı guess. I need another method instead of require(). Any suggestions?

MadLittleMods, 2018-10-01:
@halilpolathalil As I mentioned earlier, this project is not going to work on the frontend (in a browser) regardless of how you import/package it.

closed 2018-10-01 by MadLittleMods