MadLittleMods/node-usb-detection #72: Uncaught Error: a dynamic link library (dll) initialization routine failed.

hullabaloon, 2018-10-26:
I used this library on a electron app on mac without problem but on windows I get this error.
I have no idea what this is about. If someone could help, it would be great.

Here is the full error message:

```
\\?\C:\Users\mywin\Desktop\electron-app\node_modules\usb-detection\build\Release\detection.node
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (module.js:671:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)
    at Module.require (module.js:586:17)
    at require (internal/module.js:11:18)
    at bindings (C:\Users\mywin\Desktop\electron-app\node_modules\bindings\bindings.js:81:44)
    at Object.<anonymous> (C:\Users\mywin\Desktop\electron-app\node_modules\usb-detection\index.js:10:37)
    at Object.<anonymous> (C:\Users\mywin\Desktop\electron-app\node_modules\usb-detection\index.js:112:3)
    at Module._compile (module.js:642:30)
    at Object.Module._extensions..js (module.js:653:10)
    at Module.load (module.js:561:32)
    at tryModuleLoad (module.js:504:12)
    at Function.Module._load (module.js:496:3)

```

MadLittleMods, 2018-10-26:
@hullabaloon Did you already rebuild for electron? https://github.com/electron/electron/blob/master/docs/tutorial/using-native-node-modules.md

MadLittleMods, 2018-10-26:
@hullabaloon Seems like that is a problem with building `sqlite3`, not `usb-detection`

hullabaloon, 2018-10-26:
Yes, initially error pointed to usb-detection but now something else. In mac I had the same  SQLite problem but in windows solutions didn't work so far. Thanks anyway..

closed 2018-10-26 by hullabaloon