MadLittleMods/node-usb-detection #76: Create new prebuilds to include latest Electron (v69)

Kasalehlia, 2019-03-08:
Please create new prebuilds and a new release to include binaries for the latest Electron version (4.0.7) using NODE_MODULE_VERSION = 69

MadLittleMods, 2019-05-09:
Relevant code is here. Feel free to add Node.js v10 in a pull request

 - https://github.com/MadLittleMods/node-usb-detection/blob/d71db255622438f52714c9578bbf3fff3b8806ca/.travis.yml#L20-L21
 - https://github.com/MadLittleMods/node-usb-detection/blob/d71db255622438f52714c9578bbf3fff3b8806ca/appveyor.yml#L9-L10

MadLittleMods, 2019-05-09:
This seems highly related to https://github.com/MadLittleMods/node-usb-detection/issues/80

MadLittleMods, 2019-05-09:
Closing in favor of https://github.com/MadLittleMods/node-usb-detection/issues/80 as I think the electron builds come along with those updates anyway.

closed 2019-05-09 by MadLittleMods

jordao76, 2019-06-02:
Still not working for node 10:

prebuild-install WARN install No prebuilt binaries found (target=10.15.3 runtime=node arch=x64 libc= platform=win32)

Kasalehlia, 2019-07-05:
Even though #80 is closed, there are still no electron builds newer than v57. The current electron release is 5.0.0 / v70

MadLittleMods, 2019-07-05:
@Kasalehlia There is still https://github.com/MadLittleMods/node-usb-detection/issues/85