MadLittleMods/node-usb-detection #77: .on is not a function

wzr1337, 2019-04-11:
I have issues using this module in Typescript - actually in the compiled version of my code:

```bash
TypeError: usbDetect.on is not a function
    at Object.<anonymous> (/...dist/app.js:XX:YY)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Function.Module.runMain (module.js:693:10)
    at startup (bootstrap_node.js:191:16)
    at bootstrap_node.js:612:3
```

This is likely because of the use of `EventEmitter2` and not explicitly exporting the `.on()` function..

_tsconfig.json_
```
{
  "compilerOptions": {
    /* Basic Options */
    "target": "es6",                          /* Specify ECMAScript target version: 'ES3' (default), 'ES5', 'ES2015', 'ES2016', 'ES2017','ES2018' or 'ESNEXT'. */
    "module": "commonjs",                     /* Specify module code generation: 'none', 'commonjs', 'amd', 'system', 'umd', 'es2015', or 'ESNext'. */
    "lib": [
      "es6",
      "dom"
    ],                             /* Specify library files to be included in the compilation. */
    "declaration": true,                   /* Generates corresponding '.d.ts' file. */
    "baseUrl": "./src",                    /* Base directory to resolve non-absolute module names. */
    "paths": {
      "*": [
        "node_modules/@types/*"
      ]
    },                           /* A series of entries which re-map imports to lookup locations relative to the 'baseUrl'. */
    "rootDirs": ["./src", "__tests__"],       /* List of root folders whose combined content represents the structure of the project at runtime. */
    "allowSyntheticDefaultImports": true,  /* Allow default imports from modules with no default export. This does not affect code emit, just typechecking. */
    "esModuleInterop": true,                  /* Enables emit interoperability between CommonJS and ES Modules via creation of namespace objects for all imports. Implies 'allowSyntheticDefaultImports'. */
    "inlineSourceMap": true,               /* Emit a single file with source maps instead of having a separate file. */
    "skipLibCheck": true,
    "resolveJsonModule": true
  },
  "include": [
    "src/**/*"
  ],
  "exclude": [
    "node_modules",
    "dist",
  ]
}
```

_package.json_
```
{
  ...
  "dependencies": {
    ...
    "usb-detection": "^4.1.0"
  },
  "devDependencies": {
    ...
    "@types/node": "^9.4.0",
    "typescript": "^2.9.2"
  }
}
```

wzr1337, 2019-04-11:
right after posting, I found that commenting out:

```
 // "allowSyntheticDefaultImports": true,  /* Allow default imports from modules with no default export. This does not affect code emit, just typechecking. */
    // "esModuleInterop": true,                  /* Enables emit interoperability between CommonJS and ES Modules via creation of namespace objects for all imports. Implies 'allowSyntheticDefaultImports'. */
```

solves the issue

MadLittleMods, 2019-04-17:
@wzr1337 My TypeScript knowledge is very limited and I haven't used the TypeScript version of this library before. I see that `.on()` is explicitly exported here, https://github.com/MadLittleMods/node-usb-detection/blob/875c525cc55b8c8d161138e5055c6b40a15d0851/index.d.ts#L22

Is this a problem with the types defined in this project or your `tsconfig.json`?

wzr1337, 2019-04-18:
It seems to be a Type Script Compiler settings issue as stated above.. but
I am also not sure how to solve it other then switching the features off...

MadLittleMods, 2019-04-18:
I am going to close for now if it's out of this projects control. If there is a solution or best practice we should follow in this project, feel free to re-open.

closed 2019-04-18 by MadLittleMods

khuttun, 2021-04-30:
We are still seeing this issue with 4.10.0. It only appears when using the "import star" syntax (`import * as usbDetect from 'usb-detection'`), but not when using the "import default" syntax (`import usbDetect from 'usb-detection'`). To be clear, it only seems to affect the `on()` function. Other functions are found as expected. I wonder if this could have something to do with this issue: https://github.com/microsoft/TypeScript/issues/29687

khuttun, 2021-04-30:
The cause seems to be https://github.com/microsoft/TypeScript/issues/29687, as `Object.prototype.hasOwnProperty.call(require('usb-detection'), 'startMonitoring') `yields true, but `Object.prototype.hasOwnProperty.call(require('usb-detection'), 'on')` false.