MadLittleMods/node-usb-detection #78: `.find()` returns disconnected USB devices on Windows

mikew, 2019-04-16:
I've only noticed this on Windows, but it seems like `.find()` doesn't update the list of devices until the process is exited.

Repro steps:

1. Start `node` repl
1. `require('usb-detection').find().then((x) => console.log(x.length))`
1. Disconnect device
1. `require('usb-detection').find().then((x) => console.log(x.length))`
1. notice the number is the same.
1. exit `node` repl and start another one
1. `require('usb-detection').find().then((x) => console.log(x.length))`
1. notice the number is finally changed

MadLittleMods, 2019-04-17:
@mikew I just tested this on Windows 10 with a couple different devices. Starting plugged in and unplugged, and the device total changes appropriately when I disconnect/connect.

```js
setInterval(() => {
	usbDetect.find()
		.then(function(devices) {
			console.log('devices', devices.length);
		});
}, 2000);
```

 - What USB devices are you using?
 - Can you try with a different device or even PC?

MadLittleMods, 2019-04-17:
It looks like the key is to use `usbDetect.startMonitoring();`. Just tested without and the number stays the same as to what you are seeing. Feel free to submit a pull request to make this more clear in the readme.

mikew, 2019-04-17:
I was using a device that shows up as Feather M0, and noticed it in Windows 10 (both bare metal and in VMware Fusion).

I am using `startMonitoring`, but after an initial call to `find`. I can do up a PR tomorrow.

mikew, 2019-04-17:
Ah, and it's not just Windows. Happens on macOS as well.

closed 2019-04-22 by MadLittleMods