MadLittleMods/node-usb-detection #82: Cannot install usb-detection (v1.15.2): Command Failed

AshishCMSingh, 2019-05-07:
Getting the following log

`yarn add v1.15.2`
`[1/4] Resolving packages...`
`[2/4] Fetching packages...`
`info fsevents@1.2.7: The platform "win32" is incompatible with this module.`
`info "fsevents@1.2.7" is an optional dependency and failed compatibility check. Excluding it from installation.`
`[3/4] Linking dependencies...`
`[4/4] Building fresh packages...`
`[5/5] ⠐ edge-js`
`[-/5] ⠐ waiting...`
`[-/5] ⠐ waiting...`
`[4/5] ⠐ usb-detection`
`error <some path>\node_modules\usb-detection: Command failed.`
`Exit code: 1`
`Command: prebuild-install || node-gyp rebuild`
`Arguments:`
`Directory: <some path>\node_modules\usb-detection`
`Output:`
`prebuild-install WARN install No prebuilt binaries found (target=10.15.0 runtime=node arch=x64 libc= platform=win32)`

`<some path>\node_modules\usb-detection>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "" rebuild )`
`gyp info it worked if it ends with ok`
`gyp info using node-gyp@3.8.0`
`gyp info using node@10.15.0 | win32 | x64`
`gyp ERR! configure error`
`gyp ERR! stack Error: Command failed: C:\Program Files\Python37\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];`
`gyp ERR! stack   File "<string>", line 1`
`gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];`
`gyp ERR! stack                                ^`
`gyp ERR! stack SyntaxError: invalid syntax`
`gyp ERR! stack`
`gyp ERR! stack     at ChildProcess.exithandler (child_process.js:294:12)`
`gyp ERR! stack     at ChildProcess.emit (events.js:182:13)`
`gyp ERR! stack     at maybeClose (internal/child_process.js:962:16)`
`gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)`
`gyp ERR! System Windows_NT 10.0.16299`
`gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"`
`gyp ERR! cwd <some path>\node_modules\usb-detection`

MadLittleMods, 2019-05-07:
It looks like you are using Python 3.7. Probably best to **use Python 2.7 since `node-gyp` doesn't support Python 3 yet**

 - https://github.com/nodejs/node-gyp/issues/1337
 - https://github.com/nodejs/node-gyp/issues/193

AshishCMSingh, 2019-05-08:
Thank you @MadLittleMods for the quick support, by using the following command on the elevated command prompt I was able to install the _**usb-detection**_

`npm config set python C:\Python27`

closed 2019-05-08 by AshishCMSingh