MadLittleMods/node-usb-detection #85: Prebuild not found for node 10 on windows

jordao76, 2019-06-03:
Prebuild not found for node 10, on windows:

```sh
> usb-detection@4.3.0 install .\node_modules\usb-detection
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.16.0 runtime=node arch=x64 libc= platform=win32)
```

_Originally posted by @jordao76 in https://github.com/MadLittleMods/node-usb-detection/pull/81#issuecomment-498073348_

MadLittleMods, 2019-06-03:
From https://nodejs.org/en/download/releases/, I see Node.js v10 is `NODE_MODULE_VERSION: 64`

I see the [build succeeded](https://ci.appveyor.com/project/MadLittleMods/node-usb-detection/builds/24849357) but no v64 binaries [listed on that tag](https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.3.0)

Looking at the [specific build for Node.js v10](https://ci.appveyor.com/project/MadLittleMods/node-usb-detection/builds/24849357/job/w53aw85tvadij10n), I see it [installs Node.js v10](https://ci.appveyor.com/project/MadLittleMods/node-usb-detection/builds/24849357/job/w53aw85tvadij10n#L30) but it looks like something is unspecified and it [builds against Node.js 4.0.0](https://ci.appveyor.com/project/MadLittleMods/node-usb-detection/builds/24849357/job/w53aw85tvadij10n#L125) instead.

Pull request welcome if someone wants to look more into it

jordao76, 2019-06-03:
Prebuild uses node-abi for target resolution.

I do see node 4.0.0 for ABI 64 for the electron runtime here: https://github.com/lgeiger/node-abi/blob/master/index.js#L69. 

Node 10.0.0, ABI 64 for the node runtime here: https://github.com/lgeiger/node-abi/blob/master/index.js#L57

Prebuild seems to be building for the wrong runtime, electron instead of node. Seems like an issue with prebuild.

todbot, 2019-06-03:
There was a bunch of problems around the time of Node v10 and Electron v3 wrt it not reporting to node-abi the right ABI version.  The solution seemed to be "don't use Node v10".

timfish, 2019-07-04:
I've hit this issue trying to use 4.3.0 with Electron v4.2.6 and Electron builder.

I was expecting it to fall back to build locally when prebuilt binaries are not found but no luck.

MadLittleMods, 2020-04-24:
On the latest v4.8.0, I see `v64` builds https://github.com/MadLittleMods/node-usb-detection/releases which corresponds to Node.js v10.

Related context behind the update: https://github.com/MadLittleMods/node-usb-detection/pull/106

@jordao76 @todbot Does the latest v4.8.0 fix your issue?

jordao76, 2020-04-24:
Tried v4.8.0 with node v12.14.1 and it works. Sorry I won't be able to test with node 10.

PR #106 (2020-04-23): build(prebuild): update prebuild dependencies to latest
[node-abi](https://www.npmjs.com/package/node-abi), which is a dependency of prebuild and prebuild-install, contains the logic for which Node versions `prebuild --all` compiles for and subsequently publishes to GitHub releases.

`package-lock.json` had locked the version of node-abi to a version (2.2.0) from about 2 years ago, leading to a drastically outdated list of precompiled Node versions. This PR uninstalls and then reinstalls prebuild and prebuild-install to pick up the latest version of each as well as the latest version of node-a [... cut]
closed 2020-04-24 by MadLittleMods