MadLittleMods/node-usb-detection #87: was compiled against a different Node.js version using

GanZhiXiong, 2019-08-22:
[image]

> electron-quick-start@1.0.0 start /Users/ganzhixiong/Documents/electron/github/electron-quick-start
> electron .

App threw an error during load
Error: The module '/Users/ganzhixiong/Documents/electron/github/electron-quick-start/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 64. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func (electron/js2c/asar.js:155:31)
    at process.func [as dlopen] (electron/js2c/asar.js:155:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:828:18)
    at Object.func (electron/js2c/asar.js:155:31)
    at Object.func [as .node] (electron/js2c/asar.js:155:31)
    at Module.load (internal/modules/cjs/loader.js:645:32)
    at Function.Module._load (internal/modules/cjs/loader.js:560:12)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at bindings (/Users/ganzhixiong/Documents/electron/github/electron-quick-start/node_modules/bindings/bindings.js:112:48)

GanZhiXiong, 2019-08-22:
npm i electron-rebuild -s
./node_modules/.bin/electron-rebuild

Execute the above two commands to solve this problem
**However, the following error occurred again**
[image]

We need to annotate the code that USES vid
```
usbDetect.find(vid, function(err, devices) { console.log('find', devices, err); });
usbDetect.find(vid, pid, function(err, devices) { console.log('find', devices, err); });
```
**Why? Who can explain to me, thank you**

todbot, 2019-08-22:
Looks like you've not defined the variable `vid`.
You need to do that like `var vid = 0x1234` (whatever the vendorId of your device is)

MadLittleMods, 2020-04-24:
We now have `v64` (Node.js v10)  and `v73` electron prebuilds in the `usb-detection@4.8.0` , https://github.com/MadLittleMods/node-usb-detection/releases/tag/v4.8.0

Also highly related to https://github.com/MadLittleMods/node-usb-detection/issues/85

@GanZhiXiong Does `usb-detection@4.8.0`  fix your issue?

Related context behind the update: https://github.com/MadLittleMods/node-usb-detection/pull/106

closed 2020-07-19 by MadLittleMods