MadLittleMods/node-usb-detection #89: [question]The same HID device, `node-hid` and `usb-detection` get different manufacturer data

orangeChu, 2019-10-07:
This is the result of `node-hid`:
[image]

This is the result of `usb-detection`:
[image]
(note: `manufacturer: "(标准系统设备)"` means `manufacturer: "(Standard USB Host Controller)"`)

MadLittleMods, 2019-10-08:
I assume this is related to https://github.com/MadLittleMods/node-usb-detection/pull/86

cc @nononoone

orangeChu, 2019-10-08:
Thanks @MadLittleMods .
I'm afraid it's not deviceName garbled problem.
This is the result of version 4.3.0:
[image]

It looks like `node-hid` and `usb-detection` take different values.

nononoone, 2019-10-08:
> Thanks @MadLittleMods .
> I'm afraid it's not deviceName garbled problem.
> This is the result of version 4.3.0:
> [image]
> 
> It looks like `node-hid` and `usb-detection` take different values.

It was fixed in version 4.4.0

webees, 2019-10-21:
`HID`只是一种`USB`通信协议。

而`USB`有5种标准描述符：
设备描述符
配置描述符
字符描述符
接口描述符
端点描述符

`USB`描述符之间的关系是一层一层的。明显`node-usb-detection`与`node-hid`获取得描述符层次不同。

orangeChu, 2019-10-21:
> `HID`只是一种`USB`通信协议。
> 
> 而`USB`有5种标准描述符：
> 设备描述符
> 配置描述符
> 字符描述符
> 接口描述符
> 端点描述符
> 
> `USB`描述符之间的关系是一层一层的。明显`node-usb-detection`与`node-hid`获取的描述符层次不同。你可以在windows设备管理器中查看，应该与`node-usb-detection`获取的信息一致。所以`node-usb-detection`没有毛病。另外，花钱购买vid和pid的可能有你想要的结果。
> 
> This is not a bug.

Thank you for your answer. :100:

closed 2019-10-21 by orangeChu