MadLittleMods/node-usb-detection #94: Error when running in a worker thread

kryops, 2019-11-10:
When trying to run this package within a `worker_threads` worker, I get the following error when a device is added or removed:

```
FATAL ERROR: v8::HandleScope::CreateHandle() Cannot create a handle without a HandleScope
 1: 0x55f5c9b22341 node::Abort() [/usr/bin/node]
 2: 0x55f5c9b24675 node::OnFatalError(char const*, char const*) [/usr/bin/node]
 3: 0x55f5c9c89e3b v8::Utils::ReportApiFailure(char const*, char const*) [/usr/bin/node]
 4: 0x55f5c9dd94f7 v8::internal::HandleScope::Extend(v8::internal::Isolate*) [/usr/bin/node]
 5: 0x55f5c9fa26cd v8::internal::JSReceiver::GetCreationContext() [/usr/bin/node]
 6: 0x55f5c9c9f468 v8::Object::CreationContext() [/usr/bin/node]
 7: 0x55f5c9a804a1 node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*, node::async_context) [/usr/bin/node]
 8: 0x7f8a8b2ed32e NotifyRemoved(ListResultItem_t*) [/home/michael/git/vlight/node_modules/usb-detection/build/Release/detection.node]
 9: 0x7f8a8b2ee4c0  [/home/michael/git/vlight/node_modules/usb-detection/build/Release/detection.node]
10: 0x7f8a8b099630  [/usr/lib/libuv.so.1]
11: 0x7f8a8b0a9e98 uv__io_poll [/usr/lib/libuv.so.1]
12: 0x7f8a8b099fb7 uv_run [/usr/lib/libuv.so.1]
13: 0x55f5c9b6df0e node::NodeMainInstance::Run() [/usr/bin/node]
14: 0x55f5c9af152f node::Start(int, char**) [/usr/bin/node]
15: 0x7f8a8a30b153 __libc_start_main [/usr/lib/libc.so.6]
16: 0x55f5c9a7f53e _start [/usr/bin/node]
```

* Version: 4.5.0
* Node: 12.13.0
* OS: Linux 5.3.8

I tried to create a minimal reproduction: https://github.com/kryops/usb-detection-issue-worker-threads

kryops, 2019-11-11:
I also tried with the changes from https://github.com/MadLittleMods/node-usb-detection/pull/93 , but that did not make a difference either.

MadLittleMods, 2019-11-25:
I am not sure this is the use case we want to dedicate support to but happy to accept a pull request to fix it.

If you want to ensure it keeps supported, add a test for it as well. Thanks for the small reproduction already!