MadLittleMods/node-usb-detection #100: Follwing #63 can we diable debug messages like in node-usb

DomiR, 2020-01-07:
We are transporting your logs to some analytics server and these lines are causing issues in our infrastructure:

Would you be open to add a DEBUG_LOG macro like here?
https://github.com/tessel/node-usb/blob/12aee6d773fa75134a6757839d4f0e254e3d18fb/src/node_usb.h#L79

MadLittleMods, 2020-01-08:
@DomiR We should wrap the printing in a conditional `DEBUG` environment variable. I'm basing this off of https://www.npmjs.com/package/debug

macOS, Linux:
```
DEBUG=node-usb-detection npm start
```

Windows:
```
set DEBUG=node-usb-detection&&npm start
```

wdyt?

DomiR, 2020-01-08:
I’m not sure if debug supports native modules, so you would probably need to parse the debug flag and pass the flag to native. It needs to be considered, that multiple debug namespaces can be toggled by debug (e.g. DEBUG=node-usb-detection,other-lib) when parsing.

As I currently cannot see the benefit of the log, I‘m not sure if it’s worth this trouble?

MadLittleMods, 2020-01-21:
@DomiR 🤔 A compiler flag macro seems fine. Please add some instructions on how to build with and without it enabled

DomiR, 2020-01-28:
I've opened a PR, you can take a look and see how you like it :)

MadLittleMods, 2020-04-24:
Closing as https://github.com/MadLittleMods/node-usb-detection/pull/101 is now merged.

Thanks for the contribution @DomiR ❤️

closed 2020-04-24 by MadLittleMods