MadLittleMods/node-usb-detection #102: Make native code context aware

timfish, 2020-03-16:
Using `node-usb-detection` in Electron v8 I get the following warning output:

`Electron: Loading non-context-aware native module in renderer: '\\?\C:\Users\tim\Documents\Repositories\AppName\node_modules\usb-detection\build\Release\detection.node'. This is deprecated, see https://github.com/electron/electron/issues/18397.`

Details of how to fix this can be found at the [listed issue](https://github.com/electron/electron/issues/18397):
> ### What does this look like for Native Module X?
> The amount of work to become "Context Aware" will be different for every module. For native modules that act as proxies for native getters and setters -- i.e. they just expose JS apis to acccess native APIs -- it's normally as easy as adding a NAN_MODULE_WORKER_ENABLED declaration. You can find an example of that in this pull request.
>
> Some modules will need to do cleanup when the context is destroyed to ensure the module doesn't leaking memory or to cause crashes. You can add hooks to clean up after yourself with node::AddEnvironmentCleanupHook A minimal example of this can be found in this pull request.

Lan-Hekary, 2020-07-02:
Hello, 
Any news for this issue yet ??

chsing, 2021-06-30:
Hi,

Any update on this issue?

Thanks!