MadLittleMods/node-usb-detection #103: Fail to rebuild usb-detection ia32 on x64 machine

dilin-MS, 2020-03-18:
Hi, I try to rebuild usb-detection for taget arch "ia32" on x64 machine but fail. Could you please take a look?

```bash
$  node-gyp rebuild --target=7.1.11 --arch=ia32 --dist-url=https://atom.io/download/electron
gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@13.9.0 | linux | x64
gyp info find Python using Python version 3.5.2 found at "/usr/bin/python3"
gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/root/.nvm/versions/node/v13.9.0/lib/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/mnt/c/Users/dilin/workspace/repo/test/node_modules/usb-detection/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.nvm/versions/node/v13.9.0/lib/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/root/.cache/node-gyp/7.1.11/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/root/.cache/node-gyp/7.1.11',
gyp info spawn args   '-Dnode_gyp_dir=/root/.nvm/versions/node/v13.9.0/lib/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/root/.cache/node-gyp/7.1.11/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/mnt/c/Users/dilin/workspace/repo/test/node_modules/usb-detection',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/mnt/c/Users/dilin/workspace/repo/test/node_modules/usb-detection/build'
  CXX(target) Release/obj.target/detection/src/detection.o
In file included from /root/.cache/node-gyp/7.1.11/include/node/v8-internal.h:14:0,
                 from /root/.cache/node-gyp/7.1.11/include/node/v8.h:27,
                 from /root/.cache/node-gyp/7.1.11/include/node/node.h:63,
                 from ../src/detection.h:5,
                 from ../src/detection.cpp:1:
/root/.cache/node-gyp/7.1.11/include/node/v8config.h:343:49: warning: ‘MicrotasksCompletedCallback’ is deprecated [-Wdeprecated-declarations]
   declarator __attribute__((deprecated(message)))
                                                 ^
/root/.cache/node-gyp/7.1.11/include/node/v8.h:8513:3: note: in expansion of macro ‘V8_DEPRECATE_SOON’
   V8_DEPRECATE_SOON("Use *WithData version.",
   ^
/root/.cache/node-gyp/7.1.11/include/node/v8config.h:343:49: warning: ‘MicrotasksCompletedCallback’ is deprecated [-Wdeprecated-declarations]
   declarator __attribute__((deprecated(message)))
                                                 ^
/root/.cache/node-gyp/7.1.11/include/node/v8.h:8522:3: note: in expansion of macro ‘V8_DEPRECATE_SOON’
   V8_DEPRECATE_SOON("Use *WithData version.",
   ^
  CXX(target) Release/obj.target/detection/src/deviceList.o
  CXX(target) Release/obj.target/detection/src/detection_linux.o
In file included from /root/.cache/node-gyp/7.1.11/include/node/v8-internal.h:14:0,
                 from /root/.cache/node-gyp/7.1.11/include/node/v8.h:27,
                 from /root/.cache/node-gyp/7.1.11/include/node/node.h:63,
                 from ../src/detection.h:5,
                 from ../src/detection_linux.cpp:4:
/root/.cache/node-gyp/7.1.11/include/node/v8config.h:343:49: warning: ‘MicrotasksCompletedCallback’ is deprecated [-Wdeprecated-declarations]
   declarator __attribute__((deprecated(message)))
                                                 ^
/root/.cache/node-gyp/7.1.11/include/node/v8.h:8513:3: note: in expansion of macro ‘V8_DEPRECATE_SOON’
   V8_DEPRECATE_SOON("Use *WithData version.",
   ^
/root/.cache/node-gyp/7.1.11/include/node/v8config.h:343:49: warning: ‘MicrotasksCompletedCallback’ is deprecated [-Wdeprecated-declarations]
   declarator __attribute__((deprecated(message)))
                                                 ^
/root/.cache/node-gyp/7.1.11/include/node/v8.h:8522:3: note: in expansion of macro ‘V8_DEPRECATE_SOON’
   V8_DEPRECATE_SOON("Use *WithData version.",
   ^
  SOLINK_MODULE(target) Release/obj.target/detection.node
/usr/bin/ld: cannot find -ludev
collect2: error: ld returned 1 exit status
detection.target.mk:138: recipe for target 'Release/obj.target/detection.node' failed
make: *** [Release/obj.target/detection.node] Error 1
make: Leaving directory '/mnt/c/Users/dilin/workspace/repo/test/node_modules/usb-detection/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/root/.nvm/versions/node/v13.9.0/lib/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:321:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.4.0-18362-Microsoft
gyp ERR! command "/root/.nvm/versions/node/v13.9.0/bin/node" "/root/.nvm/versions/node/v13.9.0/bin/node-gyp" "rebuild" "--target=7.1.11" "--arch=ia32" "--dist-url=https://atom.io/download/electron"
gyp ERR! cwd /mnt/c/Users/dilin/workspace/repo/test/node_modules/usb-detection
gyp ERR! node -v v13.9.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok
```

environment:: WSL
```
$ uname -a
Linux cn-dilin-1210 4.4.0-18362-Microsoft #476-Microsoft Fri Nov 01 16:53:00 PST 2019 x86_64 x86_64 x86_64 GNU/Linux
```

MadLittleMods, 2020-03-18:
I have no experience building this for Electron.

Here is someone else that had troubles with `ia32` and Electron (not sure about `x64`): 

 - https://github.com/MadLittleMods/node-usb-detection/issues/88
 - https://github.com/MadLittleMods/node-usb-detection/issues/75

---

And we don't have prebuilds for that version of Electron/Node.js yet -> https://github.com/MadLittleMods/node-usb-detection/issues/85

MadLittleMods, 2020-04-24:
In `usb-detection@4.8.0`, we now have prebuilds for Node.js v12 and v13.

@dilin-MS Are you able to install successfully from a prebuild now? (just `npm install usb-detection@latest`)

dilin-MS, 2020-04-24:
Thanks!
We will try build usb-detection ia32 on x64 machine with nodejs v12 on our next release. You can safely close this issue now if you think the issue has been resolved. If I have any question later trying the latest usb-detection, I will come and comment to let you know.
Thank you very much!

closed 2020-04-24 by MadLittleMods