MadLittleMods/node-usb-detection #104: No prebuilt binaries found(target=7.1.13 runtime=electron arch=x64 libc= platform=linux)

saifeiLee, 2020-04-20:
[image]
Could we update binary prebuild package for electron-v7.1.13?

MadLittleMods, 2020-04-23:
PR welcome. Relevant prebuild code is in the CI configs:

 - https://github.com/MadLittleMods/node-usb-detection/blob/master/.travis.yml
 - https://github.com/MadLittleMods/node-usb-detection/blob/master/appveyor.yml

MadLittleMods, 2020-04-24:
@saifeiLee Does the latest v4.8.0 fix your issue?

Related context behind the update: https://github.com/MadLittleMods/node-usb-detection/pull/106

NoahAndrews, 2020-07-17:
This is now failing for me for Electron 9.0.5:
```
 • electron-builder  version=22.7.0
  • loaded configuration  file=package.json ("build" field)
  • loaded parent configuration  file=C:\Users\Noah\REV-UI\node_modules\electron-webpack\out\electron-builder.js
  • rebuilding native dependencies  dependencies=usb-detection@4.9.0 platform=win32 arch=x64
  • install prebuilt binary  name=usb-detection version=4.9.0 platform=win32 arch=x64
  • build native dependency from sources  name=usb-detection
                                          version=4.9.0
                                          platform=win32
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.5
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ C:\Users\Noah\AppData\Roaming\npm-cache\_prebuilds\b12ade-usb-detection-v4.9.0-electron-v80-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/MadLittleMods/node-usb-detection/releases/download/v4.9.0/usb-detection-v4.9.0-electron-v80-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/MadLittleMods/node-usb-detection/releases/download/v4.9.0/usb-detection-v4.9.0-electron-v80-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.0.5 runtime=electron arch=x64 libc= platform=win32)
```

MadLittleMods, 2020-07-19:
@NoahAndrews Same comments apply if you want to update things.

I see the original issue was solved from the reactions so I am going to close the issue.

PR #106 (2020-04-23): build(prebuild): update prebuild dependencies to latest
[node-abi](https://www.npmjs.com/package/node-abi), which is a dependency of prebuild and prebuild-install, contains the logic for which Node versions `prebuild --all` compiles for and subsequently publishes to GitHub releases.

`package-lock.json` had locked the version of node-abi to a version (2.2.0) from about 2 years ago, leading to a drastically outdated list of precompiled Node versions. This PR uninstalls and then reinstalls prebuild and prebuild-install to pick up the latest version of each as well as the latest version of node-a [... cut]
closed 2020-07-19 by MadLittleMods