MadLittleMods/node-usb-detection #105: Node LTS 12.16.x

zeg-io, 2020-04-21:
I've been encountering this a lot while on this project.  The module will not build.  I have the latest command-line tools on MacOS and it still blows up.

Any chance you can fix it?

```sh
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:311:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Darwin 19.4.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/tonyarcher/code/personal/slackomation/node_modules/usb-detection
gyp ERR! node -v v12.16.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm WARN angular-estree-parser@1.1.5 requires a peer of @angular/compiler@^6.0.0 || ^7.0.0 but none is installed. You must install peer dependencies yourself.
npm WARN tsutils@3.17.1 requires a peer of typescript@>=2.8.0 || >= 3.2.0-dev || >= 3.3.0-dev || >= 3.4.0-dev || >= 3.5.0-dev || >= 3.6.0-dev || >= 3.6.0-beta || >= 3.7.0-dev || >= 3.7.0-beta but none is installed. You must install peer dependencies yourself.
npm WARN slackomation@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! usb-detection@4.7.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the usb-detection@4.7.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```

MadLittleMods, 2020-04-23:
For reference, I am able to install successfully on macOS 10.15.3:

```
$ nvm use 12.16.2
Now using node v12.16.2 (npm v6.14.4)

$ node -v
v12.16.2

$ npm install usb-detection

> usb-detection@4.7.0 install /Users/eric/Downloads/usb-detection-install-test1/node_modules/usb-detection
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.16.2 runtime=node arch=x64 libc= platform=darwin)
  CXX(target) Release/obj.target/detection/src/detection.o
  CXX(target) Release/obj.target/detection/src/deviceList.o
  CXX(target) Release/obj.target/detection/src/detection_mac.o
  SOLINK_MODULE(target) Release/detection.node
npm WARN usb-detection-install-test1@1.0.0 No description
npm WARN usb-detection-install-test1@1.0.0 No repository field.

+ usb-detection@4.7.0
added 65 packages from 49 contributors and audited 101 packages in 6.481s

1 package is looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

mcous, 2020-04-23:
@MadLittleMods it appears that the lockfile has locked `[node-abi](https://www.npmjs.com/package/node-abi)` at an old version, which is causing the `prebuild` script to fail to generate prebuilt binaries for any version of Node past v9. Would you be able to fix this up? I'm personally able to build `usb-detection` on Node v12 but the lack of prebuilt binaries is proving to be a problem on CI that I'd rather not deal with (need to build on all three OSs).

I'm also happy to open a PR to bump the `prebuild` + `prebuild-install` dependencies (along with updating their nested dependencies in the lockfile), but so far I'm not able to get my local environment to behave to test if `prebuild` is still working

MadLittleMods, 2020-04-24:
@mcous Thanks for fixing it up, it looks like the latest Node.js and Electron versions have prebuilds now ❤️ 

@zeg-io Are you able to install from a prebuild now? (just `npm install usb-detection@latest`)

closed 2020-07-19 by MadLittleMods