MadLittleMods/node-usb-detection #110: compile using pkg

aufabdulnafea, 2020-05-05:
I have tried to compile a node project which uses this module using (zeit pkg). the compiled version could not work unless I put "node_modules" folder on the same path.

how could I compile the project in order to get just one standalone .exe file, as I do not want to ship the whole node_modules folder with the project?

this is the error I get: 
`pkg/prelude/bootstrap.js:1248
      throw error;
      ^

Error: Could not locate the bindings file. Tried:
 → C:\snapshot\usb_detector\node_modules\usb-detection\build\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\build\Debug\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\build\Release\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\out\Debug\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\Debug\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\out\Release\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\Release\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\build\default\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\compiled\12.13.1\win32\x64\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\addon-build\release\install-root\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\addon-build\debug\install-root\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\addon-build\default\install-root\detection.node
 → C:\snapshot\usb_detector\node_modules\usb-detection\lib\binding\node-v72-win32-x64\detection.node
    at bindings (C:\snapshot\usb_detector\node_modules\bindings\bindings.js:126:9)
    at Object.<anonymous> (C:\snapshot\usb_detector\node_modules\usb-detection\index.js:13:37)
    at Module._compile (pkg/prelude/bootstrap.js:1324:22)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:993:10)
    at Module.load (internal/modules/cjs/loader.js:813:32)
    at Function.Module._load (internal/modules/cjs/loader.js:725:14)
    at Module.require (internal/modules/cjs/loader.js:850:19)
    at Module.require (pkg/prelude/bootstrap.js:1229:31)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (C:\snapshot\usb_detector\app.js:1:75) {
  tries: [
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\build\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\build\\Debug\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\build\\Release\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\out\\Debug\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\Debug\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\out\\Release\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\Release\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\build\\default\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\compiled\\12.13.1\\win32\\x64\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\addon-build\\release\\install-root\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\addon-build\\debug\\install-root\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\addon-build\\default\\install-root\\detection.node',
    'C:\\snapshot\\usb_detector\\node_modules\\usb-detection\\lib\\binding\\node-v72-win32-x64\\detection.node'
  ]
}`

MadLittleMods, 2020-05-07:
@aufabdulnafea Not sure. Your best bet is probably to look at how to do it generally for any native C++ Node addon (`node-gyp` builds). Or maybe [`prebuild`](https://www.npmjs.com/package/prebuild) has some option.

aufabdulnafea, 2020-05-23:
I found the solution.. we just need to add the *.node file next to our executable.

closed 2020-05-23 by aufabdulnafea