MadLittleMods/node-usb-detection #116: Was compiled against a different Node.js version using NODE_MODULE_VERSION 57. This version of Node.js requires NODE_MODULE_VERSION 76.

matthew-william-lock, 2020-07-18:
I have tried to remove and use npm install as well as rebuild to fix this issue but again and again I get the same error

```
Failed to open serial port /dev/ttyUSB0 due to error: + Error: The module '/home/matthew/.vscode-insiders/extensions/vsciot-vscode.vscode-arduino-0.3.1/out/node_modules/usb-detection/build/Release/detection.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 76. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

MadLittleMods, 2020-07-19:
Which version of Node.js are you using? Feel free to create a pull request to add your version to the CI builds:

 - [`/.travis.yml`](https://github.com/MadLittleMods/node-usb-detection/blob/8188b18ea4a73c5d7801d82e72a1bf8ff8a15fe5/.travis.yml#L15-L24)
 - [`appveyor.yml`](https://github.com/MadLittleMods/node-usb-detection/blob/8188b18ea4a73c5d7801d82e72a1bf8ff8a15fe5/appveyor.yml#L6-L16)

carboleda, 2020-07-26:
@matthew-william-lock this steps work fine for me! https://github.com/serialport/node-serialport/issues/1910#issuecomment-524949720

MadLittleMods, 2020-07-27:
Thanks @carboleda! Maybe we should make a note on the readme. Would you be up to submit a PR?

 - https://github.com/MadLittleMods/node-usb-detection/issues/25#issuecomment-368014008
 - https://github.com/MadLittleMods/node-usb-detection/issues/58#issuecomment-381350676
 - https://github.com/MadLittleMods/node-usb-detection/issues/39#issuecomment-315524803

carboleda, 2020-07-28:
Hi @MadLittleMods, your welcome. Yes, in this moment I going to create a PR ;).

closed 2021-03-17 by MadLittleMods

MadLittleMods, 2021-03-17:
Added those docs via https://github.com/MadLittleMods/node-usb-detection/pull/133