MadLittleMods/node-usb-detection #123: Is it possible to detect usb devices in docker/balena container ?

adamvx, 2020-11-04:
Hi,

I tried to implement this library for my project running on balena but I was not able to make it run. It works on host but not in container even when its started with privileged flag. 

I can provide an example to try it out. 

First run the test container.
`docker run -it --name test --privileged balenalib/raspberrypi4-64-debian-node:11-buster-build bash`

Then create and new folder with `mkdir app` and inside it create new project with `npm init -y` 

Install required package with `install_packages libudev-dev` and install this library `npm i usb-detection`. This will take some time because on this specific image are no pre-builded binaries

Then create index.js `touch index.js` and paste there this snippet.
```
var usbDetect = require('usb-detection');
usbDetect.startMonitoring();

console.log('Listening...');

usbDetect.on('change', function(device) { console.log('change', device); });
usbDetect.on('change:vid', function(device) { console.log('change', device); });
usbDetect.on('change:vid:pid', function(device) { console.log('change', device); });
```

Finally run project with `node index.js` and try to connect or disconnect any usb device. I was not able to detect any device even if its present in `/dev` directory inside the container.

Does anyone has been solving similar issue or is there some guide that I am not aware of ? 

Thanks for the help.

MadLittleMods, 2020-11-04:
@adamvx Are you passing the USB devices to the container?

We won't be able to detect anything if you're just plugging and unplugging on the host without passing the USB device through.

adamvx, 2020-11-04:
@MadLittleMods Of course, all host devices are passed to container through --privileged flag. As I said, I see that device in /dev directory of the container as I see it in host /dev directory when its connected. But no event is fired.

Did you somehow managed to make it work ?

MadLittleMods, 2020-11-04:
@adamvx I test this library with linux is via a VirtualBox Ubuntu VM. The UI passthrough there is a bit clunky but the test suite does pass if I toggle the USB device through and off in the menus.

closed 2022-01-10 by MadLittleMods

iKK001, 2022-12-26:
No idea what you mean by `mounting /run/udev` ?? Can you please clarify ??

I have the exact same problem when trying to use the library with an generic-amd64 standard BalenaOS Docker image i.e. `It works on host but not in container even when its started with privileged flag`...

Here is the steps I take:

I make sure both are installed in my Docker.template file:

```
RUN sudo apt-get install -y build-essential
RUN sudo apt-get install -y libudev-dev
```

I also set the following in my Docker.template file:

```
ENV UDEV=1
```

And I make sure, the priviledge is set to true in my docker-compose.yml file:

```
privileged: true
```

But USB-detection is not working. It only works when running the code locally on my Mac. But it does not work inside the docker container.

Inside my nodejs-code, I have:

```
import usbDetect from "usb-detection"
usbDetect.startMonitoring()
usbDetect.on("add", () => {
    // ...
}
```

What exactly is still wrong ?

iKK001, 2022-12-26:
I finally found the solution:

It turned out that nothing was wrong with your `usb-detection' library!

But rather it was a needed setting in the Balena/Docker image setup.

There you need to set `network_mode: host`.

I also posted this in the Balena Forum as you can see here:

https://forums.balena.io/t/usb-re-connect-fails-from-container/363284/3?u=skuenstler