MadLittleMods/node-usb-detection #128: Why do PID and VID differ from the ones reported by `serialport`?

hrueger, 2021-02-22:
`node-usb-detection` reports this:
```
{
    locationId: 0,
    vendorId: 1155,
    productId: 22336,
    deviceName: 'Serielles USB-Gerät (COM12)',
    manufacturer: 'Microsoft',
   serialNumber: '0021003A4341570720323032',
    deviceAddress: 13
}
```
and `serialport`s `list()` method reports this:
```
{
    path: 'COM12',
    manufacturer: 'Microsoft',
    serialNumber: '0021003a4341570720323032',
    pnpId: 'USB\\VID_0483&PID_5740\\0021003A4341570720323032',
    locationId: 'Port_#0003.Hub_#0001',
    vendorId: '0483',
    productId: '5740'
}
```
The serialnumber is the same, although it is uppercase in serialport.

MadLittleMods, 2021-02-23:
@hrueger Which one is correct?

I'm not sure why they differ.

hrueger, 2021-02-23:
@MadLittleMods Thanks for the quick reply!
I'm not sure. If I do a google search for VID 0483 and PID 5740, I find the correct device: https://devicehunt.com/view/type/usb/vendor/0483/device/5740
However, searching for VID 1155 and PID 22336, I find this:
https://community.st.com/s/feed/0D50X00009XkYSESA3
On the bottom of that page, there's also something about VirtualCOMPort...

hrueger, 2021-02-23:
I think that https://www.npmjs.com/package/usb returns the same VID and PID as `node-usb-detection`, however, I don't get a serialnumber there.
Maybe your VID and PID are from the device itself and serialport's are from the serial port and not from the device itself? I don't know.

MadLittleMods, 2021-02-23:
Here are the docs for `SerialPort.list()` for reference: https://serialport.io/docs/api-stream/#serialportlist

If we are matching what `usb` returns, that seems like we're probably doing the right thing. Still would be good to understand the difference here.

@hrueger When you are using `node-usb-detection`, do you see multiple devices? Is there one listed with VID `0483` and PID `5740` as well?

You're theory could be on the right track.

MadLittleMods, 2021-02-23:
@hrueger It looks like `serialport` has their `vendorId` and `productId` in hexadecimal

 - `0x0483` =`1155`
 - `0x5740` = `22336`

hrueger, 2021-02-23:
Ah, that makes sense 👍
So I can close this issue?

MadLittleMods, 2021-02-23:
@hrueger :+1: Yep closed! Thanks for the quick follow-up here 😸

closed 2021-02-23 by MadLittleMods